logisim-evolution/logisim-evolution #5: Combinational analysis!?

JesseTG, 2015-04-19:
Where did that go?  I liked being able to switch between boolean expressions, K-maps, and truth tables; the inability to do that in Logisim Evolution is a real deal-breaker for me, even if it doesn't have much bearing on the circuit itself.

reds-heig, 2015-04-20:
Given the educational nature of Logisim, we intentionally disabled it. It is still present in the code, and you can easily reactivate it if you want (no need to recompile ;))

closed 2015-04-20 by reds-heig

JesseTG, 2015-04-21:
I still think having to enable it manually is a bit absurd.

snare, 2015-07-19:
How is this re-enabled? I don't understand why it was disabled.

reds-heig, 2015-07-20:
It can be re-enabled both from command line and from the code. It has been disabled to prevent the students to have an easy way to get Logisim do what they are supposed to do by hand.

snare, 2015-07-20:
What is the command line argument for enabling it? I don't see anything in the help output. I think disabling this by default is a bad idea. If students want to cheat and use this feature they can do so trivially by using the old version of Logisim.

snare, 2015-07-20:
Found the `-analyze` flag in the code, but it doesn't seem to work on the precompiled jar.

reds-heig, 2015-07-20:
You can find the command line options here:
https://github.com/reds-heig/logisim-evolution/blob/master/src/com/cburch/logisim/gui/start/Startup.java
The choice to disable it was not mine -- it was an explicit request from those who financially support the development.

snare, 2015-07-20:
```
java -jar logisim.jar -analyze
usage: java com.cburch.logisim.gui.start.Startup [options] [filenames]

options:
   -accents yes|no   use accented characters or ASCII equivalents
   -clearprops       clear application preferences at startup
   -empty            use empty template
   -gates shaped|rectangular   use specified gate style
   -help             display this summary and exit
   -load file        load image file into RAM (works with -tty only)
   -locale str       use locale given in str
   -nosplash         hides splash screen at startup
   -plain            use standard Logisim template
   -sub file1 file2  load file replacing library file1 with file2
   -template file    use file as template
   -tty format       run without graphical interface
   -version          display version number and exit
```

snare, 2015-07-20:
OK, fair enough.

reds-heig, 2015-07-20:
$ wget https://reds.heig-vd.ch/share/logisim-evolution/logisim-evolution-2.13.9.jar
--2015-07-20 07:32:35--  https://reds.heig-vd.ch/share/logisim-evolution/logisim-evolution-2.13.9.jar
Resolving reds.heig-vd.ch (reds.heig-vd.ch)... 193.134.221.167
Connecting to reds.heig-vd.ch (reds.heig-vd.ch)|193.134.221.167|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 34386431 (33M) [application/java-archive]
Saving to: ‘logisim-evolution-2.13.9.jar’

100%[======================================================================================================================================================================>] 34,386,431  19.7MB/s   in 1.7s   

2015-07-20 07:32:37 (19.7 MB/s) - ‘logisim-evolution-2.13.9.jar’ saved [34386431/34386431]

$ java -jar logisim-evolution-2.13.9.jar -analyze

Works fine in my case.

snare, 2015-07-20:
Ah my bad, wrong jar.

snare, 2015-07-20:
Works fine.

IanFinlayson, 2016-02-14:
I teach with Logisim and teach students how to do combinational analysis.  The first lab I have them use only transistors.  The second I have them use only gates.  And the third is using the combinational analysis tool.  Why make students do it all by hand when real designers don't?  Getting rid of this is silly IMHO.

BFH-ktt1, 2016-02-14:
What I do is that I change in:
com/cburch/logisim/Main.java the variable:
public static boolean ANALYZE = true;
recompile, and the new jar has the analyze tool enabled by default (without the need for the command line option -analyze).

jessetheasian, 2021-02-22:
If anyone is having difficulties, I found a tutorial,
https://www.youtube.com/watch?v=9p2uBG9RLqU