logisim-evolution/logisim-evolution #11: Simulation halted by internal error

lana-al, 2015-04-25:
The file contains 15 sub-ciruits, and the simulation is oftentimes interrupted by message "Simulation halted by internal error" along with several arbitrary wires changing color to either red or blue.  Sometimes issues is fixed by repeatedly restarting simulation or Ctrl+E. 
 
Stack trace printed to stdout:

	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
[... 306 lines cut ...]
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(CircuitState.java:428)
	at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Propagator.java:243)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:305)
	at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Simulator.java:126)
java.lang.NullPointerException
java.lang.NullPointerException

lana-al, 2015-04-25:
[image]

alexandremalki, 2018-02-07:
The file provided does not exists anymore. Closing the issue

closed 2018-02-07 by alexandremalki