logisim-evolution/logisim-evolution #13: Create an "export project" function

reds-heig, 2015-04-27:
It is often important to distribute a project on which one is working -- and this is particularly true for project assistants, or for students that have to hand back their work.

In these cases, one has to manually ensure that all the needed files are in the archive, and often the paths are wrong.

It would be nice to have this archive created automatically with a button directly from logisim, fixing and the same time the paths encoded in the .circ file.

maehne, 2019-01-29:
This would be really a helpful feature as I often have to deal with submitted Logisim projects by students, where one or more `.circ` file was missing. As `.circ` files are unfortunately referenced with full file system path, it is difficult to check on an own computer whether everything is self-contained in the project directory. Maybe it would already help if a relative path to included `.circ` libraries would be saved in the `.circ` file. Thus, one could check by copying/moving the project directory to a different place on one's hard drive.

BFH-ktt1, 2021-10-04:
Closing this issue, addressed by #1194

closed 2021-10-04 by BFH-ktt1