logisim-evolution/logisim-evolution #18: Include the possibility to add a new board without recompiling

reds-heig, 2015-04-27:
At present if a new card is added, before being able to see it in the list of available boards, one has to recompile and produce a new Logisim version. Ideally, this shouldn't be the case and the board should be immediately available.

BFH-ktt1, 2015-09-10:
This is already four years of my stack of TODO's. In principle there is already a menu item, it's just that the back-end fails :-< I planed for it already in the past, and hope I'll find some time in the near future to implement it, as I have seen that this would be a great improvement of flexibility. To add to this list might be the possibility to "hide" existing boards from the list or even remove them without the requirement to recompile.

BFH-ktt1, 2015-09-10:
I added 90% of this feature in my latest push. The only part that is left is to have the ability to remove external boards from the FPGA commander.

reds-heig, 2015-09-14:
Great, thanks!!

BFH-ktt1, 2017-03-31:
Completely implemented by my last push. Closing the issue.

closed 2017-03-31 by BFH-ktt1