logisim-evolution/logisim-evolution #21: Test Logisim-evolution with Modelsim

reds-heig, 2015-04-27:
It would be nice to test whether Logisim-evolution works with the free version of Modelsim, so that everyone can use advanced functionalities of Logisim-evolution.

DanHarty, 2016-04-21:
I just installed the [free student version of Modelsim](https://www.mentor.com/company/higher_ed/modelsim-student-edition) and got a simple one bit adder to work. 

Pointed Logisim to "C:\Modeltech_pe_edu_10.4a\win32pe_edu".

Configuration was slightly complicated by the Logisim documentation being for QuestaSim. Documentation seemed to hint at a /bin subfolder, which I couldn't find. I tried the highest level default installation directory "C:\Modeltech_pe_edu_10.4a" which didn't work. Finally noticed the "win32pe_edu" subfolder which contained modelsim.exe. I guess it's the actual installation directory for ModelSim itself and the highest level is for the entire ModelTech package, which is confusing because the download page was just for "ModelSim PE Student Edition". 

I don't actually know VDHL as I plan on using Logisim Evolution to learn it, so I can't do any detailed testing. The fact that a design compiles and runs at all is a _very_ good sign, though. If I do go through with using Logisim to learn VDHL, I will post any bugs I get with the free version.

If it is bug free, I'd recommend changing any references to "QuestaSim" to "QuestaSim/ModelSim Student Edition" and updating the documentation to show how to configure both.

BFH-ktt1, 2019-02-27:
Got it working without problems with Altera/Intel modelsim free edition. Closing issue for now.

closed 2019-02-27 by BFH-ktt1