logisim-evolution/logisim-evolution #23: Suppress the confusion between "Pin in" and "Pin out"

reds-heig, 2015-04-27:
People often have problems because they forget to change the in/out type of a pin, or its orientation (or both).
To solve this issue, the multi-usage pin should be replaced by two components
* Pin_In with the connection on the right and without the property "Output?"
* Pin_Out with the connection on the left and without the property "Output?"

Also, verify that the pin components in the bar are the new ones, and that this change is retro-compatible!

Akari1982, 2015-12-11:
Don't forget about bidirectional pins. In most cases on chip has data pins that works both way. It's very hacky to emulate this behavior by duplicate pins. Maybe it's better to remove this two behaviors and create one bidirectional pin.

robertorigamonti, 2015-12-11:
It is a good idea, but unfortunately I think that many parts of logisim rely on the in/out dichotomy of the pins -- having inout pins would probably create weird errors :(

BFH-ktt1, 2018-08-15:
pushed in last change, closing

closed 2018-08-15 by BFH-ktt1