logisim-evolution/logisim-evolution #25: Automatic symbol generation with I/O names (like VHDL symbols) for components

reds-heig, 2015-05-21:
A Logisim component created starting from graphical components does not show the I/O labels inside its body. Connection pins are not so visible, and one has to put the arrow exactly above it to have an information box displaying the name. This is not the case for the components created from VHDL code, which show clearly both the component and the I/O labels.

Adding this would greatly improve the lisibility of Logisim schemes.

It would also be interestin to change the position of the I/O inside the symbol. Simulink uses this trick: a property "order numeration" is added to the pins, with 1 being above the symbol, then 2, 3, ...

reds-heig, 2015-05-21:
Problem analysis (in French only):

Les modifications nécessaires ne sont pas aussi triviales que supposé initialement. Il semble très compliqué d'ajouter le nom des I/O directement au niveau des ports, car ceux-ci sont dessinés par des classes génériques relativement hautes dans la hiérarchie, et ne possèdent pas toutes les informations nécessaires. Néanmoins, il devrait être possible d'utiliser les classes permettant d'éditer l'apparence d'un circuit pour ajouter des labels directement à côté des I/O, ce qui limiterait les changements à quelques classes. En conséquence, les classes d'intérêt sont les suivantes:

    com.cburch.logisim.circuit.Circuit
    com.cburch.logisim.circuit.appear.CircuitAppearance
    com.cburch.logisim.circuit.appear.DefaultAppearance
    com.cburch.logisim.circuit.appear.PortAppearance
    com.cburch.logisim.circuit.appear.PortManager
    com.cburch.draw.shapes.Text

La classe Circuit est la classe principale des circuits. Elle contient notamment un membre "appearance", de type CircuitAppearance, qui va se charger de gérer son apparence/symbole. Celle-ci peut-être une apparence par défaut, construite dans DefaultAppearance, ou une apparence customisée lorsqu'elle est éditée par l'utilisateur via l'éditeur de composant intégré à Logisim. CircuitAppearance a donc la responsabilité de gérer et d'actualiser l'apparence du circuit auquel est elle associée.

Une première solution consisterait à modifier DefaultAppearance de telle manière à ce que l'apparence par défaut de tous les composants contienne un label associé à chaque port. Cela aurait l'avantage de s'intégrer simplement à l'éditeur d'apparence des composants. Il serait notamment possible de déplacer ces labels lors de l'édition, et de ne pas avoir à gérer leur positionnement lorsqu'on modifie l'orientation d'un composant. Pour cela, dans la classe DefaultAppearance, il faudrait:

    Modifier le rectangle du symbole dans la méthode statique build() pour avoir suffisamment de place pour les labels. On doit tenir compte du fait que des ports peuvent se situer sur les quatre côtés du symbole.
    Dans la méthode placePins(), ajouter dans 'dest' un nouveau 'Text' (une shape) qui contient le nom du port. 'dest' est une liste de canvas qui contient tous les objets graphiques qui constituent le symbole d'un composant.

     dest.add(new Text(x+offset, y, pin.getAttributeValue(StdAttr.LABEL)));

Ceci est un exemple, il faut vérifier si le label n'est pas vide, gérer l'orientation, etc. D'autre part, il faut n'appliquer les changements que pour les composants à partir de la version dans laquelle ces modifications sont effectuées. Il faut également vérifier la présence d'effets de bord, notamment dans PortManager qui fait également appel à la méthode build() de DefaultAppearance, à priori uniquement pour effectuer certaines vérifications.

Une autre solution consisterait à ajouter une option (un bouton dans la toolbar) dans l'éditeur de composant qui permettrait de faire les mêmes modifications de manière automatique (c-à-d que l'utilisateur n'aurait pas besoin de créer et positionner les labels lui-même, etc.). L'avantage de cette solution est qu'il n'y a plus besoin de vérifier la version lors du chargement d'un projet par l'utilisateur, afin de savoir quelle implémentation de DefaultAppearance utiliser.

Ces modifications, tests y compris, devraient prendre une semaine environ. Moins s'il n'y a pas de mauvaises surprises, plus si certains couplages n'ont pas été repérés. Finalement, il est peut-être également possible de modifier la classe PortAppearance afin d'y ajouter le nom du port sans passer par des labels (Text), mais la méthode draw() de cette classe ne semble pas être appelée (dans le cas d'une apparence par défaut).

BFH-ktt1, 2016-01-18:
Added by my latest push-set I think

closed 2016-01-18 by BFH-ktt1