logisim-evolution/logisim-evolution #34: Update the "User Guide"

reds-heig, 2015-05-27:
Logisim-evolution contains several new tools with respect to the original Logisim. Nonetheless, as usual :(, the documentation did not follow. The User Guide should now be updated to keep track of the news!

MarcinOrlowski, 2021-07-13:
@maehne any idea how this is looking after 6 years?

maehne, 2021-07-13:
Most improvements were contributed over the past years by @mbaillif, but not much progress has been made documenting the new features. We need volunteers for that.

maehne, 2026-08-21:
@hewzhew: Yes, very much!

maehne, 2026-09-24:
@hewzhew: Thanks for your contributions to fix this issue! Do you see other gaps, which need to be addressed? If yes, it would be good to list them here as checkbox items.