logisim-evolution/logisim-evolution #37: Show signal names in circuit's instantiation

reds-heig, 2015-05-27:
For the moment, when the mouse pointer passes over a circuit's pin, the signal name is shown. This is insufficient, as when the schema is printed these names are not visible.

Requested functionality: show in a circuit's symbol then names of the signals given in the schema.

BFH-ktt1, 2015-11-11:
I changed with my latest push the appearance of circuits such that the labels of the input(s) and outputs(s) of the circuit are directly visible in the default symbol.
Furthermore, no pins will be shown on the north and south side.
Maybe it is not the nicest display of the subcircuits, but I think it is more visible. I'm open to suggestions how we could even make the default appearance better/nicer.

reds-heig, 2015-11-12:
There seems to be a problem with the modifications: now, each time I add a component and start adding wires or pins, it creates a couple of new empty Java windows. Even changing pin's properties (like the name) seems to foster further window creation...
Could you please have a look at it?
Thanks a lot for your help! :)
[image]

BFH-ktt1, 2015-11-12:
Oops, forgot in my hack to dispose the windows ;-)
My latest push should fix this problem.
The small hack I had to make is that for getting the font metrics I need to have a graphics. As in the method the graphics is not know, I create a frame and a panel to get the font metrics. It is kind of a dirty hack, but works for now. Maybe it should be cleaned up later on.

reds-heig, 2015-11-12:
It seems to work perfectly, thanks!

closed 2016-01-20 by BFH-ktt1