logisim-evolution/logisim-evolution #45: Create a library of test circuits

reds-heig, 2015-05-27:
A test library made of circuits is needed to test that each release properly works on a standard set of example. It would be nice that those circuits include all the components, and that their output immediately reflects the success/failure state of the system (e.g., a signal that gets to 1 when everything works correctly).

reds-heig, 2015-05-27:
Some circuits could be taken from here:
http://www.learnabout-electronics.org

MosheBerman, 2015-11-10:
What about showing how some of the built in latches and other components work? My professor had us build decoders, multiplexers, and latches for one of our exams. It might be a nice thing to have too.

reds-heig, 2015-11-10:
Yes, definitely. The idea is to have something easy to run and check -- for combinatorial circuits I had added the automatic verification (taken from another Logisim version) which should do the job, but it doesn't work for sequential ones...

r0the, 2015-12-13:
You might be interested in two projects I'm currently working on:

- [LogiUnit](https://github.com/r0the/logiunit) is a unit test suite for Logisim which makes
it easy to build automated tests for circuits. The library depends on few built-in components only.
I have ported the library to logisim-evolution.

- [Logi7400](https://github.com/r0the/logi7400) is a growing library of 7400 components built from gates only (mostly). It could provide you with circuits to test against.

robertorigamonti, 2015-12-14:
That's **fantastic**!!! :)
Exactly what we were looking for. I'll give a try ASAP.
Thanks!!

MarcinOrlowski, 2021-06-30:
@BFH-ktt1 As `LogiUnit` is 404, and `Logi7400` is archived, I think this ticket can be safely closed. We need tests, for sure, This one won't help though.

closed 2021-07-01 by BFH-ktt1

jxu, 2021-07-18:
This is done with logisim test vectors right? If we still need unit tests for individual components I can help create some.

MarcinOrlowski, 2021-07-19:
We desperately needs unit tests here. If you can contribute, you are more than welcome to send your PR

jxu, 2021-07-19:
Which circuits need testing? I'm not that familiar with anything beyond basic logic gates like the 7400 series

Also should this issue be reopened or a new issue?

jxu, 2021-07-19:
Here is a simple test I did with a handwritten test vector. For more complicated circuits, it is worth using a script to auto generate test vectors.

[and_gate.zip](https://github.com/logisim-evolution/logisim-evolution/files/6843232/and_gate.zip)