logisim-evolution/logisim-evolution #49: Can't load previously saved circ

Tobias004, 2015-08-01:
File Attached:
When Adding the custom circuit named "Register" to "main", the .circ does not load. The attached file does not load. 

.circ XML:
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<project source="2.13.9" version="1.0">
This file is intended to be loaded by Logisim-evolution (https://github.com/reds-heig/logisim-evolution).
<lib desc="#Wiring" name="0">
    <tool name="Constant">
      <a name="width" val="4"/>
      <a name="value" val="0xf"/>
    </tool>
  </lib>
  <lib desc="#Gates" name="1">
    <tool name="OR Gate">
      <a name="size" val="30"/>
    </tool>
    <tool name="NOR Gate">
      <a name="size" val="30"/>
    </tool>
    <tool name="XOR Gate">
      <a name="size" val="30"/>
    </tool>
    <tool name="XNOR Gate">
      <a name="size" val="30"/>
    </tool>
  </lib>
  <lib desc="#Plexers" name="2">
    <tool name="Multiplexer">
      <a name="enable" val="false"/>
    </tool>
    <tool name="Demultiplexer">
      <a name="enable" val="false"/>
    </tool>
  </lib>
  <lib desc="#Arithmetic" name="3"/>
  <lib desc="#Memory" name="4">
    <tool name="ROM">
      <a name="contents">addr/data: 8 8
0
</a>
    </tool>
  </lib>
  <lib desc="#I/O" name="5"/>
  <lib desc="#HDL-IP" name="6">
    <tool name="VHDL Entity">
      <a name="content">--------------------------------------------------------------------------------&#13;
-- HEIG-VD, institute REDS, 1400 Yverdon-les-Bains&#13;
-- Project :&#13;
-- File    :&#13;
-- Autor   :&#13;
-- Date    :&#13;
--&#13;
--------------------------------------------------------------------------------&#13;
-- Description :&#13;
--&#13;
--------------------------------------------------------------------------------&#13;
&#13;
library ieee;&#13;
  use ieee.std_logic_1164.all;&#13;
  --use ieee.numeric_std.all;&#13;
&#13;
entity VHDL_Component is&#13;
  port(&#13;
  ------------------------------------------------------------------------------&#13;
  --Insert input ports below&#13;
    horloge_i  : in  std_logic;                    -- input bit example&#13;
    val_i      : in  std_logic_vector(3 downto 0); -- input vector example&#13;
  ------------------------------------------------------------------------------&#13;
  --Insert output ports below&#13;
[... 660 lines cut ...]
      <a name="bit4" val="1"/>
      <a name="bit5" val="1"/>
      <a name="bit6" val="1"/>
      <a name="bit7" val="1"/>
    </comp>
    <comp lib="3" loc="(270,140)" name="Comparator">
      <a name="width" val="4"/>
      <a name="mode" val="unsigned"/>
    </comp>
    <comp lib="3" loc="(580,840)" name="Shifter">
      <a name="width" val="4"/>
      <a name="shift" val="rr"/>
    </comp>
    <comp lib="3" loc="(580,690)" name="Shifter">
      <a name="width" val="4"/>
    </comp>
    <comp lib="3" loc="(580,790)" name="Shifter">
      <a name="width" val="4"/>
      <a name="shift" val="rl"/>
    </comp>
    <comp lib="0" loc="(430,160)" name="Pull Resistor">
      <a name="facing" val="north"/>
    </comp>
  </circuit>
</project>

reds-kgs, 2015-08-19:
Hello,

Did you create the circuit with another version of logisim and then open it with logisim-evolution?

There are some issues with the registers when I open the design:

at com.cburch.logisim.gui.generic.RegTabContent.getAllRegisters(Unknown Source)
at com.cburch.logisim.gui.generic.RegTabContent.fillArray(Unknown Source)
at com.cburch.logisim.gui.generic.RegTabContent.propagationCompleted(Unknown Source)
at com.cburch.logisim.circuit.Simulator.firePropagationCompleted(Unknown Source)
at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Unknown Source)

Tobias004, 2015-08-20:
Hi,
No, I created the circuit with logisim-eolution only. I don't have the original logisim installed at all.

BFH-ktt1, 2015-09-09:
Hi Tobias004,
As "Register" is already a standard component of the library memory, you cannot use this name for a sub-circuit. Try to rename in the circ-file "Register" to "Registerfile" and afterwards to open it. It should work.

Tobias004, 2015-09-09:
Thank you, it works after renaming the circuit "register" to something else.

You should definitely deny circuit Names with the same name as already known circuits!
Maybe auto correction of conflicting name(s) when saving together with a message box?

closed 2015-09-10 by BFH-ktt1