logisim-evolution/logisim-evolution #53: Check reserved names

BFH-ktt1, 2015-09-10:
Problems can occur when a sheet is saved with a name of an existing component. Also using spaces and special characters in file/directory names create problems in for example FPGA-download. The effort to build in this "student-proof" feature is for the moment unclear to me.

Tobias004, 2015-09-10:
Do you really see this program as a tool for your students only?
It's so much more.. 
Personally I don't know any good and intuitive software where you can save your work and if you want to load it on the next day, the software simply don't do it.

Now in German (sorry it's faster for me)
Das sind wenige Zeilen code (maximal eine halbe Stunde Arbeit) welche dem User viel Frust ersparen. Ich kann mir nicht vorstellen das Sie solch einen Check ernsthaft als "student-proof" Feature ansehen. Eigentlich hatte ich mir überlegt es selber einzubauen, aber das  "student-proof" ist mir wirklich eine "Spur" zu Arrogant.

BFH-ktt1, 2015-09-10:
Dear Tobias,
My "student-proof" expression is in no ways an arrogant expression but more an expression of complexity, meaning that it has to work under all circumstances. The feature is more complex than it might seem in the first place, as not only the file-names are dependent on this but also labels, and other naming conventions. Furthermore, we not only have to check on existing library component names, but also on the names in added libraries, reserved names of the back-end languages like VHDL and Verilog, and special characters that might be inserted by the user that use for example a German keyboard (like u-umlaut) which are normally not supported as characters in VHDL and Verilog. A lot of work has already be done in this direction especially in sub-circuit and label-naming, but as you see often there pop-ups a new blank, like the file naming, that we did not anticipate/found before (and in this I am very happy that you found and reported this issue!).
So in short I apologize for having maybe wrongly used the term "student-proof" and hope that you see in what sense I wanted to use it.

BFH-ktt1, 2016-01-19:
Should be resolved by my latest push

closed 2016-01-19 by BFH-ktt1