logisim-evolution/logisim-evolution #56: Add copy of components across two files

reds-heig, 2015-10-28:
Right now, the copy of components from one file to another is impossible.
It would be a nice feature to have.

MosheBerman, 2015-11-10:
Like a shared library across launches/documents?

reds-heig, 2015-11-10:
The problem with shared libraries is that you have to distribute them with the .circ file. It would be nice if one could just select a component and really copy it across instances -- provided that all dependencies are satisfied.

BFH-ktt1, 2016-01-19:
See my latest push for a partial solution

BFH-ktt1, 2016-02-01:
The current problem with logisim is that it uses it's own clipboard (hence copy and paste are not visible to the OS). Opening a new logisim window creates an independent program instance. Both logisim instances cannot "see" each other, and hence the both clipboards are not visible too them preventing from copy-paste over two logisim instances. It might be quite a challenge to provide this feature.

MosheBerman, 2016-02-01:
What about a shared cache on-disk or our own daemon that shared components?

rodolforg, 2016-02-05:
Is there any interest on using java/system clipboard?
I could try to implement that...

BFH-ktt1, 2021-09-07:
Closing as retaken in #1050

closed 2021-09-07 by BFH-ktt1