logisim-evolution/logisim-evolution #60: Power On Reset circuit in logisim; do we want it or not?

BFH-ktt1, 2015-11-22:
In real-world, all memory elements power-on with an unknown/random value. Therefore, each system has a Power-on Reset (POR) circuit to be able to put the system in a defined state. In logisim, and FPGA's this POR is completely hidden and the memory elements have on simulation start all a defined value of 0. Would it be a good idea to introduce in logisim a POR-element that would:
1) activates itself when the simulation starts, is resetted.
2) produce for a certain number of "ticks" a logic one and thereafter a logic '0'.
in combination with a default state of the flipflops/memory of eiher:
1) A random number at simulation start/reset.
2) An undefined value at simulation start/reset.
This would come more close to the behavior of ASIC-based systems.

I would appreciate your ideas

reds-heig, 2015-11-23:
If I understood it correctly, it is an interesting feature indeed.
I would not, however, alter the default behaviour of logic components --- that is, I would add in the ports an option that contains the "random" and "undefined" values at simulation start, but that defaults at the "set 0" (names should of course be better than those 3 ;)) value, so that we don't break retro-compatibility.

BFH-ktt1, 2017-03-28:
I implemented the first phase for all memory components except for the RAM. Continue in this direction

closed 2018-08-14 by BFH-ktt1