logisim-evolution/logisim-evolution #62: Valid labels are changed on original Logisim file import

r0the, 2015-12-08:
When I import an original Logisim file with VHDL compliant labels that end with an underscore, the label ist changed. For example, the label  `CLR_' ist changed to 'CLR_7d7a9dd1'.

This behaviour occurs in the latest version (v2.13.14). A minimal example is enclosed.

I don't know if this is a bug or intended behaviour, but it bugs me quite a lot, since I'm compiling a library that I'd like to keep compatible both with original Logisim and logisim-evolution. I use the trailing underscore to identify active low pins.

[evo_label_test.circ.txt](https://github.com/reds-heig/logisim-evolution/files/55169/evo_label_test.circ.txt)

robertorigamonti, 2015-12-08:
Thanks for your bug report.
This is an intended behaviour, as names ending with an underscore are not valid identifiers (at least pre-VHDL'93).
(from http://vhdl.renerta.com/mobile/source/vhd00037.htm):
"Underscores are significant characters in an identifier and basic identifiers may contain underscores, but it is not allowed to place an underscore as a first or last character of an identifier. Moreover, two underscores side by side are not allowed as well."

closed 2015-12-08 by robertorigamonti

r0the, 2015-12-08:
Thanks for your fast reply. I wasn't aware of this rule, I only looked at the formal syntax defintion which does not reflect this.