logisim-evolution/logisim-evolution #63: Create subfolders for elements and subcirquits.

Akari1982, 2015-12-11:
If project become very big with a lot of elements it's hard to organize everything using only 1 level of organization. If we can have tree of folders and subfolders with elements it will be very handy.

BFH-ktt1, 2016-01-20:
Hi Akari,

The feature is "indirectly" already provided in logisim, as you can load logisim-files as libraries. The stack can be as deep as you want. The only disadvantage is that (1) the circuits become non-modifyable, and (2) you have each time to create a new file and attach the libraries.
Hope this helps

Banyc, 2020-03-22:
I prefer the subfolder idea. If there is already a bunch of custom circuits exist in one logisim-file, no way to export it as an external library.

Besides, a simple copy-and-paste could not perfectly move a circuit since it will lose its labels.  

Also, to build a new logisim-file right next to the existing one is painful because you could not copy-and-paste the file path to save, rather it needs to navigate tons of folders to get there.

BFH-ktt1, 2021-10-04:
Closing issue as addressed by #1194. If you feel it does not resolve the problem please reopen.

closed 2021-10-04 by BFH-ktt1