logisim-evolution/logisim-evolution #64: Ability to convert Verilog code to a Logisim Circuit.

chibill, 2015-12-27:
Does this exist? If it does not how hard would it be to be created.

maehne, 2015-12-29:
Such a feature would require to implement Verilog parsing, RTL synthesis, and netlist visualization in Logisim. Even using third-party solutions (and there aren't many open-source solutions in this area with [Alliance](https://soc-extras.lip6.fr/en/alliance-abstract-en/) being a notable effort), this would require a lot of effort with IMHO little value. RTL synthesis and RTL visualization is the domain of commercial EDA tools. They are able to export gate-level VHDL/Verilog net lists. However, even parsing this reduced HDL subset is not trivial and then laying out the gates and wires in a meaningful way would still be necessary. If you look at the quite good results of commercial RTL viewers, you see that even simple digital designs quickly create complex gate-level net lists. Visualizing and simulating them in Logisim would provide little insight into their functioning.

robertorigamonti, 2016-01-04:
Maehne is unfortunately right -- it would take quite a lot of work to implement this feature, and sadly it goes well beyond our budget :(

closed 2016-01-04 by robertorigamonti

Banyc, 2020-03-04:
Consider using [yosys](https://github.com/YosysHQ/yosys) to synthesize Verilog?

quantrpeter, 2022-01-04:
i am going to create one

eatcosmos, 2024-04-04:
> i am going to create one

can open it on github?