logisim-evolution/logisim-evolution #65: "Simulation halted by internal error"

whitefieldcat, 2016-01-03:


whitefieldcat, 2016-01-03:
I use a 4K monitor and java 1.8.0_65

robertorigamonti, 2016-01-04:
Sorry to hear that :(
Could you please send us some more details --- for instance, what do you do exactly when the program crashes? Are you working on a specific file (if yes, could you please attach it to the bug report so that we can replicate the error?)
Thanks a lot for your report!

ShrewdSpirit, 2016-01-04:
I get the same error when trying to open a project. I cannot fix it by reenabling or restarting the simulation on big circuits but when I switch to other smaller circuits, it works. Here's my project file https://www.dropbox.com/s/8k7zk0b70k62wvn/Calculator_8bit.circ?dl=0

Jan 04, 2016 3:23:14 PM java.util.prefs.WindowsPreferences <init>
WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
java.lang.NullPointerException
        at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(Unknown Source)
        at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(Unknown Source)
        at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(Unknown Source)
        at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(Unknown Source)
        at com.cburch.logisim.circuit.Propagator.clearDirtyComponents(Unknown Source)
        at com.cburch.logisim.circuit.Propagator.propagate(Unknown Source)
        at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Unknown Source)
java.lang.NullPointerException
        at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(Unknown Source)
        at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(Unknown Source)
        at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(Unknown Source)
        at com.cburch.logisim.circuit.CircuitState.processDirtyPoints(Unknown Source)
        at com.cburch.logisim.circuit.Propagator.clearDirtyPoints(Unknown Source)
        at com.cburch.logisim.circuit.Propagator.propagate(Unknown Source)
        at com.cburch.logisim.circuit.Simulator$PropagationManager.run(Unknown Source)

robertorigamonti, 2016-01-04:
Concerning the "Simulation halted by internal error", it's more or less "normal": when you load the circuit, it does not have a predefined state and inputs are not yet propagated, so it stops the simulation.
To solve this issue, click on "Reset simulation" in the menu "Simulate", then check "Simulation enabled" in the same menu --- on my pc, this made the system work.
Concerning the nullpointer exceptions, here we tread in hostile territory: the internals of Logisim are a black box, I've never figured out how they work (too much documentation and probably :S), so I fear that any change there would prevent the whole system from working.
I think that, as long as the simulation works, we can leave them under the carpet...
Please let me know if this solved your problem!

ShrewdSpirit, 2016-01-04:
I do the same thing most of the times and it works (not always) Also the circuit loads without that error sometimes on my PC and not at all on my laptop which has a weaker processor. I had the same problem in the original Logisim and I thought it might be fixed in Logisim Evolution

robertorigamonti, 2016-01-04:
It strictly depends on the order on which the evaluations are performed. Since these evaluations are done in different threads, there is no way to predict which one starts (and ends) first --- it's just a matter of probability.
Changing this behaviour would mean rewriting the core of the evaluation system of Logisim, and this translates in a lot of work...

closed 2016-02-01 by robertorigamonti

RoyRoki, 2024-09-06:
Help me , i use Ubuntu 24.... , i love the losigim software  but now i face some errors.

roki@rocket:~$ logisim-evolution
Picked up _JAVA_OPTIONS: -Duser.home=/home/roki/snap/logisim-evolution/4
Gtk-Message: 01:35:28.557: Failed to load module "canberra-gtk-module"
[AWT-EventQueue-0] ERROR com.cburch.logisim.util.SocketClient - Cannot create server socket
java.net.SocketException: Permission denied
	at java.base/java.net.ServerSocket.createImpl(ServerSocket.java:320)
	at java.base/java.net.ServerSocket.getImpl(ServerSocket.java:294)
	at java.base/java.net.ServerSocket.bind(ServerSocket.java:388)
	at java.base/java.net.ServerSocket.<init>(ServerSocket.java:274)
	at java.base/java.net.ServerSocket.<init>(ServerSocket.java:167)
	at com.cburch.logisim.util.SocketClient.<init>(SocketClient.java:45)
	at com.cburch.logisim.vhdl.sim.VhdlSimulatorTop.<init>(VhdlSimulatorTop.java:47)
	at com.cburch.logisim.proj.Project.getVhdlSimulator(Project.java:370)
	at com.cburch.logisim.vhdl.gui.VhdlSimulatorConsole$VhdlSimState.stateChanged(VhdlSimulatorConsole.java:62)
	at com.cburch.logisim.vhdl.gui.VhdlSimulatorConsole.<init>(VhdlSimulatorConsole.java:99)

maehne, 2024-09-06:
@RoyRoki: Please don’t post new issues into old closed issues! You already created issue #2058, where your issue can be treated.