logisim-evolution/logisim-evolution #66: RAM input/output

rebeccaherman1, 2016-01-09:
I don't understand how the input and output data could be handled in the same wire.... Could someone perhaps post an example of using the RAM circuit?

BFH-ktt1, 2016-02-01:
Hello,

The simplest way to solve your problem is to use seperate data busses for read and write, this is an attribute of the ram.
Hope this helps.

rebeccaherman1, 2016-02-01:
It looked to me like there was only one bus, and then there was an input where we could choose in or out as the function of that wire. Perhaps I just haven't managed to find the input wire... Help me out?

BFH-ktt1, 2016-02-01:
To solve your problem there are two possibilities:
1) The hard one, you implement a tri-stated bus implementation.
2) The easy one, you use a RAM with separate read and write data buses.
for option (2) click on your RAM component and go to the window called "Properties (left bottom corner). Change the property called "Databus implementation" into "Seperate databus for read and write". You're done.

closed 2016-02-01 by BFH-ktt1