logisim-evolution/logisim-evolution #67: Window/Menu Flickering

lucasderraugh, 2016-01-09:
Everytime I place a new input down or open existing circuits, the window and menu flicker. I've narrowed it down to the commit 6e5d2f9. I'm working on a fix but just wanted to post it as an issue to see if anyone else notices this.

lucasderraugh, 2016-01-09:
So it appears that the `getWidth`, `getHeight` and `getAccent` calls are making new JFrames to calculate sizes and they're briefly visible and then invisible again. Does anyone know how to change this so that we aren't making new JFrames because this is a rather large regression.

robertorigamonti, 2016-01-11:
You are right, though the problem appears on some machines only.
Here's why these three functions pose problems (and why they were written in this way):
To calculate the size of the box, you need to get the font metrics. To get them you need a graphic pointer, which is not available at the given level, and this is why this short-cut has been adopted.

A solution to the problem could be, for instance, to determine the font metrics at start-up of logisim and make a handle with which one can read them, or to propagate the graphics handle
to the given level. Both methods require some (maybe more extensive) time to realize.

Any help is more than welcome!

BFH-ktt1, 2016-01-12:
I worked on a solution which I pushed just before.
You now have the possibility to continue using the "old" display mode or to switch to the new one. You can even combine both as I added an attribute to each circuit.
I would appreciate if you could test if the new implementation gives less problems on your system, thanks in advance!

lucasderraugh, 2016-01-12:
How do you switch to the old mode, I still get the flickering on the current build.

BFH-ktt1, 2016-01-13:
To switch to the "old" mode you have to untick window->preferences->layout->"Use pin labels on circuit boxes"

lucasderraugh, 2016-01-13:
Yes, that does resolve the issue. Could this become the default perhaps?

closed 2016-01-20 by BFH-ktt1