logisim-evolution/logisim-evolution #69: Logisim Evolution and 4K resolution

ypaul123, 2016-01-15:
Hey guys,

I'm trying to run logisim evolution on my laptop which has a 4k monitor, and the menu items look really really small..

how can I make them look bigger without down scaling the resolution please ??

thanks in advance for your help

BFH-ktt1, 2016-01-20:
Hi,
Unfortunately scaling of the fonts is for the moment not integrated in logisim-evolution.
Implementing this feature would be a nice add-on

maehne, 2016-01-28:
I found the following links dealing with support of HiDPI displays in Java. Maybe they can give a hint how to tackle this issue in the Logisim implementation:
* [Retina support in Oracle JDK 1.7.0_40 and above](http://bulenkov.com/2013/06/23/retina-support-in-oracle-jdk-1-7/)
* [Resolution Independent GUIs. How to Do It, and How It Looks.](http://www.javalobby.org/java/forums/t101878.html)
* [HiDPI with Eclipse and NetBeans](https://jaxenter.com/netbeans/hidpi-with-eclipse-and-netbeans)
* [JEP 263: HiDPI Graphics on Windows and Linux](http://openjdk.java.net/jeps/263)
* [Help! Java Scaling issues on HiDPI Screen, Linux](https://www.reddit.com/r/javahelp/comments/390uxy/help_java_scaling_issues_on_hidpi_screen_linux/)
* [Fix scaling of java-based applications for a high DPI screen](http://askubuntu.com/questions/526484/fix-scaling-of-java-based-applications-for-a-high-dpi-screen)

samdolt, 2016-02-29:
On Linux, if we force GTKLookAndFeel ( Native look and feel for linux) with:
```sh
java -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel  -jar ${HOME}/.local/share/logisim/logisim.jar
```
the main interface with menu and text is properly scaled up.

Unfortunaly, in the diagram area, text are scaled up, but not the graphics.

With Gtk:
[image]

Without:
[image]

Here is the documentation for settings the look and feel in Java Swing: https://docs.oracle.com/javase/tutorial/uiswing/lookandfeel/plaf.html

samdolt, 2016-02-29:
I have just noticed that custom block render correct with Gtk:

[image]

maehne, 2016-11-13:
A student of mine, Jonas Messerli, pointed me to the following link to work around this issue on at least Windows systems:

http://superuser.com/questions/988379/how-do-i-run-java-apps-upscaled-on-a-high-dpi-display

Basically, the manifests of ```java.exe``` and ```javaw.exe``` on Windows claim that the application is DPI-aware even though a Swing-based GUI is not scaled and therefore becomes so tiny on High-DPI screens that the user interface gets to the point of becoming unusable. Previously, it would suffice to pass ```-Dsun.java2d.dpiaware=false``` to the JRE when launching the .jar file. Unfortunately, the flag does not anymore work due to the following JDK issue [JDK-8080153](https://bugs.openjdk.java.net/browse/JDK-8080153). Therefore, the manifest of the ```java.exe``` and ```javaw.exe``` used to execute the ```logisim-evolution.jar``` needs to be modified as described in the first link using, e.g., the [Ressource Tuner](http://www.restuner.com/) tool. As this might, influence other Java applications, this manifest patch should be done on a JRE version embedded into a Logisim-evolution executable wrapper. [Launch4j](http://launch4j.sourceforge.net) is one tool which is able to create such an executable wrapper around a JAR file.

roy77, 2017-02-10:
I have implemented a slider for font scaling. It works for menu, component tree, dialogs (after reopening),...
Issues:
- Toolbar and some texts are not scaled.
- Pins Value in curcuit view is mistakenly scaled.

I had problems with git, therefore the changes are in to commits.
[https://github.com/roy77/logisim-evolution/commit/ef4fced5aa6b3edd75412566d9a26e943ea58e77](url)
[https://github.com/roy77/logisim-evolution/commit/2d1f362d620e67b20421ac89bc633c8fa79a629a](url)

[image]
[image]

BFH-ktt1, 2017-03-08:
I implemented something similar as roy77 (thanks for your ideas!) that scales as well the icons as the text. Currently not everything is scaled yet (like help, FPGA-commander, Register viewer, and plenty others), but the main working area is scaled. Please give it a try and let me know your feed-back.

BFH-ktt1, 2017-03-12:
In the current scaled-wip branch there is a complete scaled implementation. Please try it out and give me some feed-back

alexandremalki, 2017-03-15:
Hello,
This week I am going to give it a try and let you know.
Thanks

maehne, 2017-03-15:
I have tested the scaled-wip branch @63e9433 on a MacBook Air late 2011 running OS X 10.11.6 x86_64 compiling and executing Logisim-evolution with the JDK version 1.8.0_121. The scaling works in most places as expected and will be certainly helpful for people working with Logisim-evolution on Hi-DPI screens. However, there are still some annoying display artifacts at higher scaling factors, as you can see in the attached screenshot, which uses a scale factor of 2.0:

[image]

As you can see the hight of the entries in the library pane is not adjusted to the font hight so that the entries are clipped. Also, the tab bar widgets is not scaled to adjust to the font height.

IMHO, the scale factor for the GUI elements should be set by default to a value, which ensures that the GUI elements match the sizes of the equivalent elements in native applications on this platform. Determining this factor in a cross-platform manor might be difficult if Java doesn't provide an API for this.

maehne, 2017-03-19:
I just tested the updated the branch scaling_resolution_wip with PR #114 included. The scaling issues on OS X described in my last comment above persist. Certain GUI elements such as pane selectors and drop-down lists aren't scaled to fit the scaled labels in them.

Additional scaling problems remain in the FPGA commander and its dialog "Component to FPGA board mapping": The windows have a fixed size and can become bigger than the screen, which renders them unusable.

maehne, 2017-03-20:
According to [Apple's Java Documentation](https://developer.apple.com/library/content/documentation/Java/Conceptual/Java14Development/04-JavaUIToolkits/JavaUIToolkits.html), the Aqua Look and Feel apparently limits widgets heights of ```JComboBox``` and ```JTabbedPane``` on purpose, which would explain the observed behavior. Switching to Swing's cross-platform look and feel might help.

maehne, 2017-03-20:
Yes, using the cross-platform look and feel (set by @BFH-ktt1 in commit 5e33325) helps to fix the widget scaling issue. As this makes the application to look more foreign to the user's desktop environment, it would be nice to be able to switch between the system's native look and feel and the cross-platform look and feel by adding an option to the preference window tab.

BFH-ktt1, 2017-03-27:
Done

maehne, 2017-04-02:
Thanks @BFH-ktt1  for implementing the configuration option option for setting the look-and-feel! I successfully tested it on OS X and Linux using the branch at 9530292.

miho, 2017-05-04:
FYI: to solve resolution issues on OS X, just add 

```xml
	<key>NSHighResolutionCapable</key>
	<string>True</string>
```
before the end of the line that starts with `</dict>` to the `Info.plist` of the application bundle. Text as well as parts and wires are drawn in high resolution now.

maehne, 2017-05-10:
Thanks for this hint. However, this key is already present in the ```Info.plist``` of the application bundle generated by the ant package target. The scaling issues have been mostly resolved by @BFH-ktt1 in the ```scaling_resolution_wip``` branch.

closed 2017-07-27 by BFH-ktt1