logisim-evolution/logisim-evolution #74: Error generation SRG component

etienne-messerli, 2016-02-02:
The Shift Register component  is functioning correctly in simulation mode. But we get a error when we try to  run the integration (FPGA commander menu). 

I give the messages we get during "Download" commande into FPGA Commander menu.
Log Error: 
   1> ***FATAL*** Failed to optimize (AREA) Project, cannot download
Log Console: 
  Error (10500): VHDL syntax error at Shift_Register_behavior.vhd(25) near text "IF";  expecting ";" File: /home/messerli/logisim_workspace/Logisim_Problemes/Bloc_Shift_Register/vhdl/memory/Shift_Register_behavior.vhd Line: 25

BFH-ktt1, 2016-02-02:
Could you please privide a logisim circuit (simple) to reproduce the bug?

etienne-messerli, 2016-02-02:
[Logisim_Problemes_circ.zip](https://github.com/reds-heig/logisim-evolution/files/114152/Logisim_Problemes_circ.zip)

commit a6e0872d (2016-02-02): Bugfix fixes #74
closed 2016-02-02 by BFH-ktt1