logisim-evolution/logisim-evolution #76: Port I/O and Reptar

BFH-ktt1, 2016-02-03:
Currently in the netlist generation there are two components that are excluded from "correct" generation. There are some issues with this:
Reptar: I've notices that this component has a correct input/output behavior. Is the exclusion in the netlist generator an "old ghost" that can be removed?

Port I/O: This component has for the moment an inout behavior and a complex way of generating the HDL-code. Due to the input-output dichomy in HDL some severe problems can occur when source and sinks are not specified for a net. To be able to generate a "good" netlist the behavior of the different pins of the Port I/O need to be provided. I see a possible solution in adding attributes to each of the pins specifying it's behavior (similar to the inversion attributes on the inputs of a gate). For the moment being I opt to exclude this component from netlist generation (= none supported component) until a clean solution is found.

robertorigamonti, 2016-02-04:
The Port I/O is used by all the board we have, its exclusion would have catastrophic effects on us!

robertorigamonti, 2016-02-04:
Could you please pin-point us where in the code this generation is performed? We could in this way investigate it a bit and give you a more detailed answer...

BFH-ktt1, 2016-02-04:
The "problem" lies in the generation of the netlist in com.bfh.logisim.designrulecheck.netlist.java
To be able to process "complex wiring" a wire coming out of a splitter and going into another one needs to know where is the source and where is the sink. The current netlist generator has still a bug that causes a connection like for example (not inherently related to the port I/O):
s_LOGISIM_NET_6  <= s_LOGISIM_BUS_1(0);
s_LOGISIM_NET_6  <= s_LOGISIM_BUS_40(0);
where it should be:
s_LOGISIM_NET_6  <= s_LOGISIM_BUS_1(0);
s_LOGISIM_BUS_40(0)  <= s_LOGISIM_NET_6;
This due to incorrect/missing source/sink annotations.
In my new rewrite of the netlist generation I want to be more explicit on source sink annotations to be able to detect netlist generation bugs like the one shown above (it does not happen often, but when it happens there is no means of telling how it influences the behavior on the FPGA).
Therefore I propose to replace the Port I/O by two possible replacements:
1) Fixed Input/Output behavior: The user has to specify for each pin of the port I/O if it is an input or if it is an output.
2) IO-block: Each "external connection" has 3 logisim pins, namely, input, output, and tristate enable. In this way we "emulate" a simple FPGA IOB.

closed 2018-10-20 by BFH-ktt1