logisim-evolution/logisim-evolution #79: Support for dualport RAM

Akari1982, 2016-02-09:
It will be nice to have option to set address separately for read and write. This type of memory widely used in PSX CPU and i can't simulate it via other elements ((

https://en.wikipedia.org/wiki/Dual-ported_RAM

http://wiki.psxdev.ru/images/f/ff/Unit01_overview.jpg

francocurotto, 2016-10-12:
I concur.

electrodeyt, 2017-06-04:
I agree

BFH-ktt1, 2017-07-26:
I agree also and will put it on my to-do list. The issue that needs to be solved here is:
1) The behavior in case the read- and write happen on the same address (read after write or write after read)
2) The behavior in case two writes happen to the same address.
Dual ported memories normally have two seperated read- and write ports. For a memory that has a seperated read and seperated write port are more often reffered to as register files.

trayres, 2021-07-21:
Hopefully this helps someone else - if you're looking to do a simple dual port style RAM, and clocked behavior is good enough (i.e. not a true dual port SRAM):

It looks like you can model one by using a RAMCLK that is 2x the SYSCLK, then simply take the first cycle to write both memories, and the second cycle to do both reads. This is a "write-first" style simple dual port style RAM:

[image]

That might fit your simulation needs.

MarcinOrlowski, 2021-07-21:
You may want to attach ZIPed *circ file. Also I'd recommend posting this in Discussions

trayres, 2021-07-22:
Sorry, I put it here because it was at the top of the Google results so I thought it would be useful.
Moved to discussion here:
https://github.com/logisim-evolution/logisim-evolution/discussions/838

File and test bench located here:
https://github.com/trayres/logisim-evolution-examples/blob/main/sdp_export.circ

maehne, 2021-07-22:
@BFH-ktt1: Was this issue accidentally closed? Your overhauls of the RAM component in PR #312 and PR #315 don't seem to have added the capability to configure the RAM component as dual-port RAM with independent addresses per port.

primechaser, 2022-02-01:
@Akari1982 - perhaps you want to try the example I have attached. I took the HDL code generated by Logisim (based on a specific example - 8 bit address, 32bits data) and just added a read ADR line to it.
The "oe" line now provides the ability to toggle between write ADR and read ADR for retrieving data, where 

"0": data based on ADR_read and
"1": data based on ADR_write

hope this helps

[dualportRAM.zip](https://github.com/logisim-evolution/logisim-evolution/files/7978151/dualportRAM.zip)

maehne, 2022-02-02:
Thanks @primechaser for sharing your solution. It shows nicely the capability of Logisim-evolution to integrate sub-circuits, which are described in VHDL. However, it will require the installation of ModelSim/QuestaSim for simulation. Also, generics are not considered at the moment. So, it is a good start, but improvements in this area are still needed.

prlaba, 2022-05-23:
In my case I need a "modified" dual-port RAM, with one set of address lines for reading and another set for writing.  I only need to read from one cell at a time, and write to one cell at a time.  But I must be able to read from one cell (gate onto a data bus) and write to the same cell (from the data bus) in the same clock cycle (I can use low-to-high CLK transition for read, and high-to-low transition of CLK for write).

The Digital logic simulator has dual-ported RAM components, would be nice if L-E did too.  I'm not a Java programmer so I'm not able to create my own native components in L-E.