logisim-evolution/logisim-evolution #84: Test Ghdl support

roy77, 2016-03-22:
Hi,

I replaced Questa Advanced Simulator with GHDL to get a VHDL programming environment with open source components.

Link to my fork: https://github.com/roy77/logisim-evolution

It works. Only inout signals are currently not supported.

Is this from interest for you? Maybe we can discuss  what I have to change that you pull my changes.

alexandremalki, 2016-03-23:
Hi roy77,
Thank you very much for your work on Logisim. 
What you did is interesting as GHDL is the leading open source vhdl simulator. 
As it is still limited ( we do not have graphical viewer so we cant see the waves), would it be possible to integrate GHDL in logisim, keep Questa Advanced simulator and let the user choose which simulator one wants to use (e.g In Preferences menu the user can switch in between 
Questa or GHDL)   ?

Once this changes are performed do a pull request we would be happy to integrate your code :).

harwoodr, 2017-08-17:
Did anything ever come of this?

roy77, 2017-08-17:
Sorry, not from me. I analyzed the code last year and there were extensive changes necessary. Before add such a feature it make sense to optimize the structure. The vhdl validation and the simulation code should base on the same classes before adding a second simulator. That was one of the reasons for me to switch to [digital](https://github.com/hneemann/Digital).

Naheel-Azawy, 2021-08-15:
Digital is nice but logisim-evolution has some other features that I find interesting. @roy77 can you please share your fork? It seems that it has been deleted

maehne, 2021-08-25:
I also would like to see a revival of this effort to support co-simulation with [GHDL](https://github.com/ghdl/ghdl). GHDL is development has been very intense over the past years. Its conformance to the VHDL IEEE standard is very good and its [documentation](https://ghdl.github.io/ghdl/) has been very much improved -- also in the field of [co-simulation](https://ghdl.github.io/ghdl-cosim).

I think it is important to offer the user possibility to choose the HDL simulator. In other regions, Verilog is more popular so other people might be interested to have support added for, e.g., [Icarus Verilog](http://iverilog.icarus.com/) or [Verilator](https://www.veripool.org/verilator/) as well.