logisim-evolution/logisim-evolution #87: Tunnel labels lost during copy/paste

DanHarty, 2016-04-22:
Tunnels lose their label when they are copy/pasted. They retain all other attributes like font, direction, and bit width.

roy77, 2017-02-09:
In my current project I need many connected tunnels and therefore have the same name/label. But with copy/paste the name/label is lost.

For a better behavior I disable the clearing of the label for class tunnel.

6d7ddbb

A better solution could be a class member which hold information about clearing.

alexandremalki, 2017-02-10:
Hi roy77,
Thank you for your help, indeed that was very annoying to write again and again the label each time you copy them (and by nature you need at least two with the same name).

We will investigate on how to make it cleaner.

BFH-ktt1, 2017-03-28:
This issue should have been solved in the two main branches. Closing the issue.

closed 2017-03-28 by BFH-ktt1