logisim-evolution/logisim-evolution #88: Frustrating Labels

DanHarty, 2016-04-22:
Required HDL compliant labels and subcircuit names makes Logisim-Evolution much more frustrating to use than Logisim, especially for newly migrated users like myself.

Is it really only VHDL simulation that requires HDL labels? I suggest an entirely separate, project wide "VHDL mode" which when initiated resolves all duplicate labels with a dialogue. Alternatively this dialogue could appear whenever the VHDL simulation box is checked and there are duplicate labels.

VHDL simulation should be targeted at users looking to graduate from the intuitive nature of Logisim to the more advanced and real-world applicable VDHL, but who want to stick with Logisim's interface. These users are likely actively looking to avoid dealing with Questa Advanced Simulator or similar software. Anyone experienced enough to know VDHL and have Questa Advanced Simulator isn't going to switch to Logisim-Evolution.

Naming scheme:
The renaming system would append "_subcircuit_#", with the number based on the labels coordinates, priority upper left. For example if two labels share the name "A_in" then they would be renamed "A_in_0" and "A_in_1".

Alternative idea:
If labeling must be consistent (no "VHDL mode" labeling), I think the ideal situation would be having labels as merely a visual thing like in Logisim.  A new "HDL compliant label" attribute could be added which the HDL features could use. It would be auto-generated based on the label, and editable.

Triavanicus, 2020-07-15:
Something else that can be done is if the user enters a character that isn't compliant with the HDL names, you could replace it with the hex value of the character.

BFH-ktt1, 2020-07-15:
The solution that @kevinawalsh has taken is also nice, instead of using the labels in vhdl, use the hierarchical name with the position in the sheet. Disadvantage, the HDL becomes even less readable. Possible solution would be an intermediate version that the label is taken if it is unique and valid, otherwise the hierarchical name with position. Still for me on a very low priority.

BFH-ktt1, 2026-09-01:
Closing this issue as it has been solved.

closed 2026-09-01 by BFH-ktt1