logisim-evolution/logisim-evolution #90: [VHDL generation] ucf file is not created

disaderp, 2016-05-21:
I tried to generate both VHDL and Verilog files from my circuit. However the ucf folder was empty for both modes(VHDL and Verilog generation). 

At first I thought it would not be a problem but when I tried to manually create NETs. Although logisim is kinda confusing all connections and it's hard to keep track at more complex project: 
```
   s_LOGISIM_BUS_1(3) <= LOGISIM_INPUT_BUBBLES(0);
   s_LOGISIM_BUS_1(1) <= LOGISIM_INPUT_BUBBLES(1);
   LOGISIM_OUTPUT_BUBBLES(0) <= s_LOGISIM_BUS_2(1);
   LOGISIM_OUTPUT_BUBBLES(1) <= s_LOGISIM_BUS_2(0);
   s_LOGISIM_BUS_1(0) <= LOGISIM_INPUT_BUBBLES(2);
   LOGISIM_OUTPUT_BUBBLES(2) <= s_LOGISIM_BUS_2(3);
```

In conclusion: without the generation of ucf file it is very hard to use the generated code

BFH-ktt1, 2017-03-08:
For which board you try this, a logisim supported, or a not supported? When the board is supported click on the button on top of the FPGA-commander, and a UCF is generated. If your board is not supported, you have to make yourself a board description file with the board-editor before logisim is capable to generate a ucf-file.

disaderp, 2017-03-09:
For every board I am trying(from a dropdown list), the button 'Generate HDL only' is disabled

BFH-ktt1, 2017-03-28:
You have to have the Third-party software installed, e.g. Xilinx-ISE for Xilinx-FPGA-based boards and Altera/Intel Quartus for Altera/Intel-FPGA-based boards. You than have to select in the FPGA-commander with the button "toolchain" the path to the tools, e.g. <path>/quartus/bin64 where the binaries of the tools are located. Afterwards the button "Generate HDL only" will change in "Download to board" and your design is automatically downloaded to your attached board (read: ucf-generated; P&R done; bitfile-generation done; download by JTAG).
Hope this helps

closed 2017-03-29 by BFH-ktt1