logisim-evolution/logisim-evolution #95: Proposal: Import circuit schematics from Digital Works

AlmuHS, 2016-09-12:
Hi all:
I want to ask if would be possible implement a new functionality in Logisim.
This functionality would allow import circuit schematics from Digital Works.

At this moment, Logisim only includes schematics of generic components.
I need to use real schematics of integrated circuits in my simulation, which I have in Digital Works format

If I could import this from Digital Works, I could work with Logisim without problems.

It would be possible to include this functionality?

AlmuHS, 2016-09-12:
I add example of Digital Works schematics 

[Integrated Circuits.zip](https://github.com/reds-heig/logisim-evolution/files/467780/Integrated.Circuits.zip)

BFH-ktt1, 2016-12-14:
The importation of Digital Works format is not foreseen in the near future. It is a nice idea, but our (little) resources are focused first on other topics. Please feel free to add this feature update logisims source tree.

closed 2016-12-14 by BFH-ktt1

AlmuHS, 2016-12-14:
I comprends
In this case, could be possible add a real schematics library in Logisim?
Create from zero, not imported