logisim-evolution/logisim-evolution #96: Book link is broken

jvalleroy, 2016-11-19:
In README.md:
"Another good reference is this book, where basic electronics is explained with the help of Logisim."

The link goes to http://www.lulu.com/shop/george-self/exploring-digital-logic-with-logisim-ebook/ebook/product-21118223.html, which says "can't find this product".

maehne, 2017-03-09:
This seems to be the correct link to the paperback edition: http://www.lulu.com/shop/george-self/exploring-digital-logic/paperback/product-22747579.html
An e-book version doesn't seem to exist anymore.

egrath, 2017-12-12:
The Book and the accompanying Lab Manual are released under the Terms of "CC BY 3.0" License, so it should be allowed to share them according to it. Eventually this project should host them here for anyone interested; I wrote the author to get a copy of them for my own purposes and have them available when needed.

georgeself, 2018-02-23:
I'm working on a completely new book with enhanced labs and better descriptions. I expect that to be complete by about July 1, 2018. I'll post a link here when it's done.

grself, 2019-02-13:
I just now noticed this discussion -- sorry about the long lag time. Here are my Github links for the digital logic book and accompanying lab manual:

[Digital Logic Text](https://github.com/grself/CIS221_Text)
[Logisim-evolution Lab Manual](https://github.com/grself/CIS221_Lab_Manual)

I'm updating both for the upcoming fall term (2019) but they should be available at my Github repository as I make changes.

BFH-ktt1, 2020-03-08:
Should be fixed, closing issue

closed 2020-03-08 by BFH-ktt1