logisim-evolution/logisim-evolution #98: Real Integrated Circuits schematics library

AlmuHS, 2016-12-14:
Hi other time:

As my previous proposal was closed, I want to do a new proposal
Instead add Digital Works libraries import support, I think that could be interesting to add a new library with schematics of differents IC real models (as 74HC86)

This could be used in simulations, allowing set connections as a circuit component, and test your input/output.

Could be possible add this new library?

maehne, 2016-12-15:
Do the libraries available from the following links meet your purposes?
* http://www.cburch.com/logisim/links.html
* http://74x.weebly.com/blog/library-of-7400-logic-for-logisim

AlmuHS, 2016-12-15:
‎This can be util, but I also need other families as I refered before.                                                                                                                                                                                   Thanks

maehne, 2016-12-15:
You know that you can build your own Logisim libraries by drawing the internal schematics of those ICs using the basic logic gates provided by Logisim. Then, you just need to edit the symbol for that sub-circuit to resemble the IC housing. You may consider to share the results of this work so others may profit of this effort as well. It is unlikely that others will start working on such a library if they don't have an immediate use case. (Many users of Logisim-evolution target FPGAs nowadays.)

AlmuHS, 2016-12-15:
‎I didn't know this.  I'll try to build it.‎Thanks by the info

closed 2017-03-29 by BFH-ktt1