logisim-evolution/logisim-evolution #100: Subcircuits with RAM share RAM state

carlor, 2016-12-18:
Let's say you have a circuit `main`, and a circuit `subcircuit` which has RAM inside it.

If `main` has two `subcircuit`s inside it, though e.g. registers in the `subcircuit`s can have separate values, if one `subcircuit` writes to RAM, the value will appear in the other RAM. I don't think that's intended.

I've attached an example.
[ram_bug.zip](https://github.com/reds-heig/logisim-evolution/files/659576/ram_bug.zip)

BFH-ktt1, 2017-03-28:
Dear Carlor,
Sorry for replying so late! You are right, the ram has a serious flaw. I tracked down the problem which is related to the Attribute Ram.CONTENTS_ATTR; in the implementation of the RAM's all which have the same other attributes share this particular attribute which contains the RAM-contents. Cloning this attribute will solve the problem; however, it will break the ability to change the RAM-contents by the attribute editor. I'm investigating further in this issue as I am currently implementing the possibility that all memory components start with an "unknown" state as in real circuits. Please stay tuned and keep an eye on the current scaling-wip branch on which I'm working.

BFH-ktt1, 2017-03-29:
My latest push in the scaling-resolution wip fixes the problem, please give it a try

closed 2017-03-29 by BFH-ktt1