logisim-evolution/logisim-evolution #105: BUG: Adding new pins rotating labels + subcircuit rotation

BFH-ktt1, 2017-02-19:
There is an issue introduced by this nice feature:
If I have modified slightly the appearance of the sub-circuit, the labels etc. will show up duplicated (see attached logisim circuit).

[example.zip](https://github.com/reds-heig/logisim-evolution/files/785798/example.zip)

alexandremalki, 2017-02-20:
Hi Theo,
I reverted this commit. You can pull the last commit. Thank you for your feedback. I am going to work on solving the issue on the next release.

BFH-ktt1, 2017-03-03:
Hi Alexandre,
Unfortunately the revert broke completely the copy-past feature of logisim. When selecting a block, performing a copy, and than a paste will only paste the wires. All components are lost in the pasted block.

alexandremalki, 2017-03-03:
This is correct in the newest comit.

BFH-ktt1, 2017-03-08:
Super, Thanks!

closed 2017-03-08 by BFH-ktt1