logisim-evolution/logisim-evolution #107: Cannot run logisim since commit  90be82af1f085da048c76b7e13c11e7005d81e19

alexandremalki, 2017-03-10:
Hi Theo, I saw you added GUI graphical scale. 
But it broke Logisim. I cannot run it anymore. 

Culprit commit : 90be82af1f085da048c76b7e13c11e7005d81e19

This is the output of my ant run :
run:
     [java] Exception in thread "main" java.lang.ExceptionInInitializerError
     [java] 	at com.cburch.logisim.util.LocaleManager.<clinit>(Unknown Source)
     [java] 	at com.cburch.logisim.gui.start.Startup.parseArgs(Unknown Source)
     [java] 	at com.cburch.logisim.Main.main(Unknown Source)
     [java] Caused by: java.lang.RuntimeException: No locale bundles are available
     [java] 	at com.cburch.logisim.util.LocaleManager.loadDefault(Unknown Source)
     [java] 	at com.cburch.logisim.util.LocaleManager.<init>(Unknown Source)
     [java] 	at com.cburch.logisim.util.Strings.<clinit>(Unknown Source)
     [java] 	... 3 more
     [java] Java Result: 1

Best regards

alexandremalki, 2017-03-10:
Hi Theo.
I created a new branch from where you can add the new scaling feature. Now the commit 214a3687c638b9d05634bad8af785af5b751c577 revert the commits : 

90be82af1f085da048c76b7e13c11e7005d81e19
a9fa92cd64efa97fc28ae7c98d8948b730703bb2
560cddf696a3be768435ac22f8c902ba56fc396f

They are still available on another branch called scaling_resolution_wip. You/I will merge it once it's working.

Thank you for your work.
Best

BFH-ktt1, 2017-03-10:
Hi Alexandre,
The error you had, had probably nothing to do with the latest commit, but with an eclipse issue, It cannot find the locales. To fix this problem just run the eclipse-fix script. I had this problem already often with eclipse, it sometimes screws things up.
This effect gives exactly the problem that logisim is not able to find the locales:
[java] Caused by: java.lang.RuntimeException: No locale bundles are available
Could you check and undo the branch, as I already added a lot more that I did not push yet and are still working on the master branch.

alexandremalki, 2017-03-10:
Hi Theo,
We tries with other machine here, and the problem was not eclipse related since we run using ant.
So i guess something is wrong somewhere else no?

Thank you

BFH-ktt1, 2017-03-10:
Yep, you are right and not ;-)
I found the problem:
When using eclipse and ant, the original build script causes a duplication of the boards in the FPGA-commander. Therefore a made following changes to the build.xml script:
- <fileset dir="." includes="doc/" excludes="doc/**/html/guide/analyze/" />
+            <fileset dir="." includes="doc/ resources/" excludes="doc/**/html/guide/analyze/" />
Changing this line back (hence adding recources/ to your includes) will solve the problem.
Sorry for committing my build.xml script (with all the files that I committed I forgot to exclude the build.xml.
This should solve your problem.
Regards,
T.
Could you branch back?

alexandremalki, 2017-03-10:
Regarding branching back. This release might impact a lot of our legacy designs etc...
What do you think about using the branch I created until all laboratory's design are tested?
If you really want to use the master i will branch back... But U.I changes impacts could be huge (especially with logisim)....

If you decide to keep using the scaling_resolution_wip for testing and push it on the github server please do as follow:

git checkout -b scaling_resolution_wip_local
git checkout scaling_resolution_wip
git merge scaling_resolution_wip_local

And to push on the branch scaling_resolution_wip
git push scaling_resolution_wip

Then to align the master on your side with the server's master:
git reset 214a3687c638b9d05634bad8af785af5b751c577 --hard

This will remove changes on your local and align the remote master with your local.

Best regards

This should do the trick.

BFH-ktt1, 2017-03-11:
Fixed the build.xml-file for the small "problem" with eclipse and duplicated board-files.
Pushed the latest scaled version of logisim in the branch.
I will continue on the branch and let you know when I'm finished

BFH-ktt1, 2017-03-12:
My changes are in. Pleace check and merge with the main branch.

closed 2017-03-12 by BFH-ktt1