logisim-evolution/logisim-evolution #108: OS X: java process keeps running after closing the last Logisim-evolution window

maehne, 2017-03-15:
This issue seems to be OS-X-specific: If Logisim-evolution is quit using the menu File -> Quit, it quits correctly and also the java process executing the logisim-evolution.jar finishes properly. However, if you try to close Logisim-evolution by simply closing the last Logisim-evolution window using the "red pill" on the top left of the window, the application continues running, i.e., the Logisim-evolution menu is still visible, but none of the Logisim-evolution menu entries works anymore -- even File -> Quit. Only quitting the java process by selecting the menu item "Main -> Quit Main" itself works.

On OS X, it is untypical that the File menu contains an item to end a program. This menu item should be only present in the application menu ("Main" in the case of the unbundled logisim-evolution.jar).

I have observed this behavior on OS X 10.11.6 with JDK 1.8.0_121. I noticed this behavior already at least a year ago and it is still present in Logisim-evolution @63e9433.

alexandremalki, 2018-02-21:
Hi @maehne ,
We mainly focus our development under Linux. If you have the solution to this problem, or find someone who has the solution, please feel free to point us to the patch.

Thank you

maehne, 2018-02-21:
I unfortunately don't have a patch and not enough time to dive myself deep into the Java implementation. Therefore, a hint for where to start looking would be appreciated. From my point of view, the callback which handles the clean-up after a window is closed has to check whether there are any more Logisim-evolution windows open. If this is not the case, the callback has to invoke the same actions, which are invoked by the callback triggered by the "File -> Quit" menu item.

davidhutchens, 2018-06-10:
I offer a solution to this and other problems on Mac OS X. This solution also fixes the issue that double clicking a .circ file in the Finder isn't launching the app and opening the file as expected.

The implementation removes MRJAdapter.jar from the project. This library is no longer maintainable. It heavily relies on libraries that are deprecated in Java 9 and even removed entirely from the JDK. So MRJAdapter cannot be compiled after Java 8. But the functionality it used to provide is either not needed, or better implemented with new features of the Java 9 Desktop classes.

The implementation replaces the two Mac specific files in Logisim-evolution so they use Desktop.

I am compiling in Java 10 (Java 9 should work, but not earlier versions). I have tested it on Mac OS 10.13.5 and on Linux 4.14.10-1-ARCH (where I didn't notice any changes).

Check my macDesktop branch for the implementation.

I invite comments or suggestions about any regression or improvement.

maehne, 2019-01-16:
> Can I suggest that this is common and acceptable behavior for a Mac app? Please refer to https://www.howtogeek.com/279086/why-do-mac-apps-stay-open-when-i-hit-the-red-x-button/ for questions.

@brianmcfadden: I am aware of this behaviour on macOS. However, Logisim-evolution is not working properly anymore after the last window got closed using its red X button. Even though the menu remains visible, none of the menu items are working anymore. So, you cannot create a new schematic or open an existing file. Nothing happens for any of the menu items. So this is clearly a bug requiring fixing! Unfortunately, I had a very busy term so that I did not yet find the time to check out the solution proposed by @davidhutchens. Due to the recent migration to Gradle and other fixes to the code base, rebasing his modifications may now require some more work.

davidhutchens, 2019-01-20:
@maehne: I used my solution with version 2.14.2 teaching last semester. It worked well. I ran into an issue with a bug in the Java run-time when trying to move to a later version and did not have time to figure out what was happening.

Since it requires compiling with Java 9 or later, I assumed you wished to maintain Java 8 compatibility and were not yet interested. But if you'd like me to try to get it working with the latest master, I'm willing to work on it. Let me know. If you'd like the Mac app (rather than having to build it with ant) just to see how it behaves, I can get that to you.

davidhutchens, 2019-01-22:
You can find the app here: http://cs.millersville.edu/~dhutchens/Logisim/

Since my semester is starting this week and I'll need to upgrade my build environment, integrate my changes, and test the result, this will likely take some time. If you have further advice or comments, they are welcome.

maehne, 2019-01-22:
@davidhutchens: Thanks a lot for providing access to your Mac app of Logisim-evolution! It keeps its promise to integrate better with macOS and fixing this issue. So, we should definitely rebase and integrate your modifications into the `develop` branch.

Concerning the build environment, I manage mine using [MacPorts](
https://www.macports.org/), through which I installed `openjdk11` and `gradle`.
```bash
$ sudo port -v selfupdate
$ sudo port -v install openjdk11 gradle
```
When you are in the `logisim-evolution` project root directory, it is sufficient to launch `gradle jar` to build a JAR archive and `gradle run` to launch the application. `gradle tasks` will show all tasks runnable from the root project. There is no equivalence to `ant package` yet (issue #192). So, no app bundle can be currently built. I started to look into this. New tasks need to be added to the gradle build scripts:

* I found the [`javafx-gradle-plugin`](
https://github.com/FibreFoX/javafx-gradle-plugin), which unfortunately relies on `javapackager`, which was removed in JDK 11. Therefore, it doesn't seem a future-proof option (`javafx-gradle-plugin` issue [#103](https://github.com/FibreFoX/javafx-gradle-plugin/issues/103)).

* There's the [Gradle Setup Builder Plugin](https://github.com/i-net-software/SetupBuilder), which seems to solve the part of creating MSI, DEB, RPM, and DMG packages.

* There's the [gradle-macappbundle](https://github.com/crotwell/gradle-macappbundle/wiki/Intro) plugin for creating macOS .app applications.

* There's an article ["The magic of Gradle: create Windows installers, Debian packages, manage a PPA, and optional sub-projects"](https://blog.heckel.xyz/2014/05/29/gradle-create-windows-installers-debian-packages-manage-a-ppa-and-optional-sub-projects/) by Philipp C. Heckel, which may contain some more useful hints.

I have not yet found the time to look at them further and try integrating them into our Gradle build scripts. Give me a note if things have settled down on your side to work on the issue so that we can avoid doubling the work. Once I find some more time to look closer at this issue, I will create a branch for it in my personal fork of the Logisim-evolution repository and create a PR to track progress on it. I will then invite you to join the discussion/effort to fix the issue.

davidhutchens, 2019-01-22:
The removal of javapackager from JDK 11 is unfortunate. I will, of course, need to build a Mac app to test anything. I looked around a bit this morning. Hopefully, the situation will settle soon. I did find this link [Filling the Packager gap](https://mail.openjdk.java.net/pipermail/openjfx-dev/2018-September/022500.html) that might be useful. [Using the Java Packager with JDK 11](https://medium.com/@adam_carroll/java-packager-with-jdk11-31b3d620f4a8) suggests it can work for the Mac and mentions gradle.

maehne, 2019-01-23:
@davidhutchens: Please check out my PR #214, which incorporates your (rebased) patches.

maehne, 2019-01-25:
Fixed by PR #214

closed 2019-01-25 by maehne