logisim-evolution/logisim-evolution #110: FPGA Commander Pin Assignment Window can become larger than the user's screen

maehne, 2017-03-15:
The FPGA Commander Pin Assignment Window can become larger than the user's screen if the zoom feature is used. Then the dialog becomes more or less unusable and Logisim-evolution needs to be restarted to recover from this problem and permitting the continuation of the work with a risk of data loss due to the forced quitting of the application.

alexandremalki, 2017-03-16:
Hi Maehne,
Thanks for your input, this is an issue i already faced and started to work on.

Best

alexandremalki, 2017-03-16:
The error is fixed in the commit mentionned above.
Thank you

BFH-ktt1, 2017-03-17:
Hallo Torsten,
It has been fixed in the latest version in the scaling-wip-branch for the new layout of both board-editor and fpga-commander. The tool will not allow to scale bigger than the screen-size.

maehne, 2017-03-19:
Dear Alexandre and Theo,

thanks for addressing so quickly my bug report! 

@alexandremalki: your fix in d8fcaef resets does not prevent the dialog from becoming to big when hitting the scale button in the component map dialog. Though, closing and reopening the dialog helps to get a usable mapping dialog.

@BFH-ktt1: The scaling slider in the mapping dialog is a usability improvement. The dialog's size is limited. Though, this is at least not reflected on the slider's position after moving it with the mouse. Only after clicking again in the slider widget, the slider moves to the position that reflects the current limited zoom size.

Personally, I find the zoom slider still a bit unintuitive. I would prefer that both the FPGA Commander dialog and the Component Mapping dialog would be both resizable windows, which would adjust their content according to the window size.

alexandremalki, 2018-02-21:
Hi Torsten,
Under which OS are you working on? 

Thank you

maehne, 2018-02-21:
Hello Alexandre,

I am working primarily on macOS (currently version 10.13.3) and when using the FPGA Commander on Linux (Debian x86_64).