logisim-evolution/logisim-evolution #111: Combinatorial analysis missing

Leumass96, 2017-03-16:
In the experimental version 2.13.23, the combinatorial analysis is missing. This functionality is very useful so it is a big loss.

Thanks

maehne, 2017-03-16:
The combinatorial analysis is always deactivated by default due to didactic reasons. It can be enabled by launching logisim-evolution.jar with the -analyze option. The default can be set via the static member variable ANALYZE in the Main class.

BFH-ktt1, 2017-03-28:
You can now also build with ant jar_all which will generate a jar with the analyzer enabled and the documentation included.

closed 2017-03-28 by BFH-ktt1

Leumass96, 2017-04-04:
I am sorry but I can't find how to export correctly the java project into a jar file which I can run. Can you help me ?

leadrien, 2017-04-04:
You need to use apache ant command.
In the project root directory run: 
`ant jar`
It will produce a logisim-evolution.jar file

maehne, 2017-04-04:
If you are compiling from the ```scaling_resolution_wip``` branch with the most recent changes, you may also try in addition ```ant package```, which will generate an OS-specific installer with logisim-evolution properly wrapped into an executable, which facilitates associating the ```.circ``` file extension for opening up in logisim-evolution.

Leumass96, 2017-04-04:
Thank you very much for your help ! I did it and I didn't know ant. It is very useful.
Thanks

jessetheasian, 2021-02-22:
If anyone is having trouble enabling the 'analysis circuit' tool I found a tutorial: https://www.youtube.com/watch?v=9p2uBG9RLqU