logisim-evolution/logisim-evolution #119: RAM with unlatched output

cogwheel, 2017-04-23:
I'd like to be able to set a RAM module to have its output always follow its address line (the way ROMs behave). This is necessary to implement a single-cycle CPU.

BFH-ktt1, 2017-04-24:
Dear cogwheel,

You can achieve this functionality by either taking an asynchronous RAM (trigger -> High level) or to use a synchronous one that clocks on the negative edge.

Hope this helps

jennselby, 2019-12-05:
I ran into the same problem when using this in a class in which I was teaching students how to build a CPU.

Unfortunately, those suggestions from BFH-ktt1 do not get the functionality we need, because we still need the writes to be edge triggered. It is only the read that we want to be asynchronous. The trigger can only be set for both.

We are using software workarounds, making sure to use two cycles for RAM load operations (that is, writing RAM load operations twice in our assembly code), so that the correct value is on the bus in the second cycle. Any operations that would involve loading RAM contents for an ALU operation that also uses a register as both input and output have to be avoided.

rkoe, 2019-12-19:
Wouldn't it be sensible to use **separate triggers** for read and write?
i.e.: Read RAM at rising edge, write at falling edge?

It would be great if this could be implemented.

I know that I could add some external gates to achieve [*] this (and also what @jennselby needs), but it would be much nicer if the RAM would already support this.

[*] e.g. for read-at-rising-edge, write-at-falling edge:
- RAM: Trigger=Rising Edge
- [image]
- [image]

BFH-ktt1, 2020-01-07:
Implemented in my latest push

closed 2020-01-07 by BFH-ktt1