logisim-evolution/logisim-evolution #121: VHDL - Counter s_carry not clocked

NassimBounouas, 2017-06-22:
Hi, 
I'm using logisim-evolution and an Altera DE2 board.
I built a simple circuit with three 4 bits counters and 3 seven segments to count from 0x0 to 0xFFF.
I'm using a Button as clock for the first counter and the carry is linked on the second seven segment clock to propagate.

The displayed count was : 0x00d -> 0x01e -> 0x02f -> 0x020
After reading the generated counter in VHDL it seems that the s_carry is not clocked. 
Because of that the counter value and the carry value aren't synchronized.

I fixed this problem by moving the instruction : 
`CompareOut   <= s_carry; `
after the instruction : 
`ELSIF (GlobalClock'event AND (GlobalClock = temp(0))) THEN`

I moved it directly into the generated VHDL file but it can be done directly in logisim-evolution source in the file package com.cburch.logisim.std.memory.CounterHDLGeneratorFactory.java (https://github.com/reds-heig/logisim-evolution/blob/39708375fd58a2d40a1324743f60b531e8995029/src/com/cburch/logisim/std/memory/CounterHDLGeneratorFactory.java) by moving the line 85 after the line 148.

maehne, 2017-06-22:
From your description it sounds more like your counter is asynchronous due to the use of the carry signal as clock for the second counter. However, the carry should be rather used as an enable signal for the next digit and all counters should be attached to the same clock signal.

The problems linked to asynchronous behavior are usually exposed once the designed is synthesized and loaded into an FPGA. Therefore, I don't think a fix to the counter VHDL code generation is necessary.

If you think that there is still a bug in Logisim-evolution, then please provide a minimal test case in form of a circuit file.

NassimBounouas, 2017-06-23:
Hi maehne,

Yes my counter was intentionally asynchronous. I modified my design to use all the counters synchronously and the behavior has been fixed. My misunderstood was coming from the overflow notion. In my mind the carry was supposed to be triggered on an overflow not as a signal on the maximum value.

Thank you for the explanation !

alexandremalki, 2018-02-21:
It seems like the issue is solved. I am going to close the issue

closed 2018-02-21 by alexandremalki