logisim-evolution/logisim-evolution #122: RAM — VHDL has  2 cycles latency whereas simulation shows 1 cycle latency

piernov, 2017-06-30:
Hi,
When simulating in Logisim, the RAM shows a one cycle latency behaviour. ie. after selecting the address and pulling either `Output Enable` or `Write Enable` high, the operation is processed on the next edge on the `Clock` input.

However, when generating VHDL and deploying on a FPGA, the RAM now shows a 2 cycles latency. 2 edges on the `Clock` input are needed to process both read and write operation.

Is this a bug? How do I get a one cycle latency behaviour in the generated VHDL?

piernov, 2017-06-30:
It sounds like the issue comes from the input/output buffers `s_ram_data_out`, `s_DataInReg`, `s_Address_reg`, or something like that. If I pass the data from `DataIn` directly to `s_mem_contents` and from `s_mem_contents` to ` DataOut`, it seems to work in one cycle.
See dirty workaround at: MIAOU-Polytech/logisim-evolution@9504c358d7dc6ce8596ad64ecae1712df118b8ae

BFH-ktt1, 2017-07-26:
The behavior of the memory is related to real implementations of RAM components. There is always two ways how a read combined with a write to the same address is handled:
1) Write after Read: In this case the read data will be the old value of the memory location and the write happens after the read: It apears that the read has 2 cycle latency (as you observed in the VHDL). This is the most common implementation of memory components.
2) Read after Write: In this case the write will be performed first, and than the read is performed. In this case you will observe a single-cycle latency.
Currently the VHDL-implementation has the Write after Read behavior which is, and I agree here with you, not consistent with the logisim behavior, and hence a "bug". I'll put it on my to-do list.

alexandremalki, 2018-02-21:
Hti @BFH-ktt1 ,
Is this issue solved? If not, maybe you shall consider leting the user to choose the behaviour. Otherwise, we can just take the patch from plernov and close the issue.
Thank you.

BFH-ktt1, 2018-08-14:
VHDL now has same behavior as logisim component, namely write-before-read

commit af7ab91a (2018-08-14): Bug fix #122
closed 2018-08-14 by BFH-ktt1