logisim-evolution/logisim-evolution #124: Inappropriate minimizing in Combinational Analysis

yakoh, 2017-07-21:
Hi,
I sometime  teach designing logic circuits using Logisim.  I found that
Logisim outputs an inappropriate expression in Combinational Analysis.
I gave the following Table.  Here, A, B, and C are inputs and X is an output.
A	B	C	X
0	0	0	0
0	0	1	1
0	1	0	1
0	1	1	1
1	0	0	x
1	0	1	x
1	1	0	1
1	1	1	0
The expected expression is X=~A C + B ~C .
However, Logisim output X= ~A C + ~A B + A ~C, which is not a minimal one.
Is it a bug?  Or, are there any reasons underlying the solution?

BFH-ktt1, 2017-07-26:
I acknowledge your findings. As this part of the code is of the original logism 2.7 version it might be that in this particular case the used algorithm does not select the minimal form (due to one of the don't cares in the table). I put it on the todo-list to look into it.
Thanks for the clear test-case.

BFH-ktt1, 2018-10-12:
The issue should have been resolved for most cases in my latest push

closed 2018-10-12 by BFH-ktt1