logisim-evolution/logisim-evolution #126: Infinite message loop when renaming

matmanj, 2017-08-06:
Begin by copying a component.  (e.g. Register)
The copy's label will be a duplicate of the original.
Change the label of the copy.
If the new label chosen is also a duplicate of an existing label a message box shows and advises to chose a unique label.
When the message is closed Logisim changes the label back to it's previous value, which is also a duplicate.
Result is an unbreakable loop of message boxes from which I've found no recovery except to kill the Java process and lose all unsaved changes. 

I have reproduced this in 2.13.22 and 2.14.1.

BFH-ktt1, 2017-08-17:
I reproduced this effect only using the duplicate function of logisim as the copy function will already, by default, erase the label. 
Will look into a fix.

BFH-ktt1, 2017-08-21:
Fixed in latest push

closed 2017-08-21 by BFH-ktt1

alexandremalki, 2017-08-23:
Copy and paste is not impacted indeed only the duplicate or CTRL + D  was impacted.
Tested patched and validated on my side. I will merge in the master.