logisim-evolution/logisim-evolution #127: Continually prompting for update even when fully updated

thomasklash, 2017-09-17:
The program continually prompts to update on launch, no matter what:
[image]

alexandremalki, 2017-09-19:
Hi,
It should be fixed by now. Would you please retry and let me know the status?
Thank you.
Have a nice day.

thomasklash, 2017-09-20:
Still happens 
[image]

alexandremalki, 2017-09-20:
This is weird. Here on our side the problème is resolved.
So now please could you:

- You try to  update your logisim by pulling the last version? 
- Press yes. If the problem continues go then press no, and let me know your version.

Best regards

thomasklash, 2017-09-20:
It goes away after pressing "Yes", but it should not be appearing at all

alexandremalki, 2017-09-20:
What about you press no? This way i can just check the version you are running

thomasklash, 2017-09-20:
I pressed no, I am running version 2.14.2, which I mentioned earlier :) latest commit compiled

AerodynamicBrick, 2017-09-20:
I'm having the same issue. Keeps asking for update to .2 I'm on .1 tho?
Won't update

alexandremalki, 2017-09-20:
Could you try to clean the directory (ant clean) and then rebuild it (ant build) and run (ant run) ?

Thank you

alexandremalki, 2017-09-20:
Also, on what platform are you running?

AerodynamicBrick, 2017-09-20:
Win10

thomasklash, 2017-09-20:
I'm on Win7 x64

alexandremalki, 2017-09-21:
Hi there,
I was thinking that maybe you on the wrong branch. Are you pulling from master?

alexandremalki, 2017-09-21:
Ok from the source, oould you please checkout
git checkout /src/com/cburch/logisim/Main.java ?

If this does not work, could you check the version number in the file aforementioned?

thomasklash, 2017-09-21:
After `ant clean` and `ant run` the new compiled `.jar` no longer has this issue.

closed 2017-09-21 by thomasklash

alexandremalki, 2017-09-21:
Good, happy it works