logisim-evolution/logisim-evolution #129: Multiplier issues.

LordDecapo, 2017-09-27:
I am using the precompiled version 2.14.2 on win10.
I was trying to make a 32 bit multiplier and unlike every other part that has a max bitwidth of 32, the multiplier seems to be capped at 31bits.
[https://i.imgur.com/mpntkn6.png](url)

alexandremalki, 2017-09-28:
Hello,
The issue here is that the output limitation is 32 bits in logisim. So the limitation on logisim should be of 16 bits not 31 bits on the inputs. Since the output size can possibly be of sizeof(input1) + sizeof(input2). 

IMHO, i think the input of the  multiplier should be lowered to 16 bits.

LordDecapo, 2017-09-28:
Yes, that is the output limitation. But in original Logisim, if you placed
a 32bit multiplier, you would simply get the lower 32bits out of the
output, and the upper 32bits out of the carry. This works great if I set
the multiplier to 31 bits, I get 31bits out of both the output and the
carry.
But it won't let me go up 1 more bit to the logisim max.

Like in my edit, I was able to copy paste a 32bit multiplier from an old
design and it functions perfect. I was thinking maybe just a typo in some
code related to the multiplier config.

alexandremalki, 2017-09-28:
Ok then, I ll take a look at this issue later. 
As for now, I leave this ticket open. 

Have a nice day

LordDecapo, 2017-09-28:
Awesome, thanks!

BFH-ktt1, 2017-09-28:
There is here an issue concerning interpretation of the bits:
The current implementation of the multiplier in java uses 32-bit integers (hence 2's complement interpretation of the bits) to calculate the result. The multiplier, however, has a behavior of binairy interpretation of the bits (hence unsigned numbers). The result of this original implementation of the multiplier was an "unsigned" multiplication upto 31 bits of inputs and "only" a "signed" multiplication for 32 bits. This is why I decided at a certain time to restrict the multiplier to 31 bits, as in this way we have a consistent way of the definition of the muliplication, namly an "unsigned" version. To make the whole thing consistent we should consider an extra flag on the multiplier to indicate if a signed (2's complement) or unsigned (binairy) multiplication should be performed, which is still on my "to do" list.
The current implementation of logisim is now consistent; meaning the multiplierr implements an "unsigned" multiplication.
Extending to 32 bits can be done, but requires some rethinking and remoddeling.
These are my small inputs on this issue ;-)

YulienPohl, 2018-03-08:
I have solved the bug with a pseudo-unsigned long cast if a 32-bit multiplier is used.

Example:
```
int a = 0x80000000;
int b = 0x00000001;
int c_in = 0;
long p1 = (a & 0xffffffffL) * (b & 0xffffffffL) + (c_in & 0xffffffffL); // fix
long p2 = (long)a * (long)b + (long)c_in; // current
// p1 = 0x0000000080000000;
// p2 = 0xffffffff80000000;
//        | carry |  out  |
```

BFH-ktt1, 2018-10-17:
Guys, are these merges a solution to this issue (read can I close this issue)?

closed 2018-10-21 by BFH-ktt1