logisim-evolution/logisim-evolution #130: Zooming behavior is strange

metabuddy, 2017-09-28:
Hi,

It seems possible only to zoom into upper-left corner and then pan the project with arrows, which is kind of awkward and strange.
Could you please implement zooming into any region of the project and add some sort of panning capability, so we can easier navigate the project.

Thank you

LordDecapo, 2017-09-28:
I second this panning feature

BFH-ktt1, 2017-09-28:
This sound like a great feature, please implement it and send us a merge request.

LogisimIt, 2018-03-28:
I did it a few months ago in my fork, if you want, just copy it

BFH-ktt1, 2018-04-03:
Could you please submit a pull-request on these changes such that I can merge it into the testing branch?
Thanks for the work!

BFH-ktt1, 2018-08-04:
New zoom behavior based on the logisim-ita branch has been implemented in the scaling_resolution_wip branch, hence closing this issue,

closed 2018-08-04 by BFH-ktt1