logisim-evolution/logisim-evolution #131: Crash while starting with Java 9 (macOS)

mnme, 2017-10-03:
Neither the app package nor the jar work.

Command line output
```
$ java -jar logisim-evolution.jar
Exception in thread "main" java.lang.ExceptionInInitializerError
	at com.cburch.logisim.gui.start.MacOsAdapter.addListeners(Unknown Source)
	at com.cburch.logisim.gui.start.Startup.registerHandler(Unknown Source)
	at com.cburch.logisim.gui.start.Startup.parseArgs(Unknown Source)
	at com.cburch.logisim.Main.main(Unknown Source)
Caused by: java.lang.StringIndexOutOfBoundsException: begin 0, end 3, length 1
	at java.base/java.lang.String.checkBoundsBeginEnd(String.java:3116)
	at java.base/java.lang.String.substring(String.java:1885)
	at net.roydesign.mac.MRJAdapter.<clinit>(MRJAdapter.java:181)
	... 4 more
```

Java version output
```
java version "9"
Java(TM) SE Runtime Environment (build 9+181)
Java HotSpot(TM) 64-Bit Server VM (build 9+181, mixed mode)
```

HELLSCORPION, 2017-10-15:
Starting with Java 9 they changed how the version information is displayed
System.getProperty("java.version") used to return stuff like
1.8.0_144
Now it just returns 9 for me
A dependency of theirs, which is not mentioned in the stacktrace, but packaged in the jar does this:
`   static {
      String prop = System.getProperty("java.version");
      javaVersion = new Float(prop.substring(0, 3));`
And logically the String index 2 is now no longer accessible, as the String is a single digit
So there is your crash and its cause: Some dependency used an interesting method to figure out the java version and was punished

Error found by @I-Al-Istannen

marvk, 2017-10-16:
I fixed the bug in the related library (mrjadapter) and submitted a pull request.

alexandremalki, 2017-10-16:
The patch was merge on the master branch. Thank you.

marvk, 2017-10-16:
I think this is not fixed in the current state. I've submitted a PR to the libraray that is causing the issue but it has not been merged with the library yet. You merged an unrelated PR.

alexandremalki, 2017-10-17:
Ticket reopened.

alexandremalki, 2017-10-17:
Just a quick copy and paste from the other ticket:

b1u3-YumaNoguchi commented 15 hours ago 

Hi!

I found a issue about java9.
When logisim-evolution run with java 9, it doesn't work.
The cause is net.roydesign.mac.MRJAdapter.java in mrjadapter.jar.

Then I got code from krichter722/arjadapter
and changed net.roydesign.mac.MRJAdapter.java

from

    javaVersion = new Float(prop.substring(0, 3)).floatValue();

to

    if(Integer.valueOf(String.valueOf(prop.charAt(0)))>=9){
    javaVersion = new Float(prop);
    }else{
    javaVersion = new Float(prop.substring(0, 3)).floatValue();
    }

I compiled it and built logisim-evolution. It ran well.
But this is my first issue. Is this right? And How can I do?

leadrien, 2017-10-17:
Fixed in 59b610b6f59762ed395b2306ae62ebd703309346

closed 2017-10-17 by leadrien