logisim-evolution/logisim-evolution #133: Java 9 issue

b1u3-YumaNoguchi, 2017-10-16:
Hi!

I found a issue about java9.
When logisim-evolution run with java 9, it doesn't work.
The cause is net.roydesign.mac.MRJAdapter.java in mrjadapter.jar.

Then I got code from [krichter722/arjadapter](https://github.com/krichter722/mrjadapter/tree/master/mrjadapter/src/main/java/net/roydesign)
and changed net.roydesign.mac.MRJAdapter.java

from

>javaVersion = new Float(prop.substring(0, 3)).floatValue();

to 

> if(Integer.valueOf(String.valueOf(prop.charAt(0)))>=9){
>                 javaVersion = new Float(prop);
>         }else{
>                 javaVersion = new Float(prop.substring(0, 3)).floatValue();
> }

I compiled it and built logisim-evolution. It ran well.
But this is my first issue. Is this right? And How can I do?

marvk, 2017-10-16:
This [issue exists already](https://github.com/reds-heig/logisim-evolution/issues/131), but has been (apparently) wrongly closed:

b1u3-YumaNoguchi, 2017-10-16:
Oh,  you reopened that issue. I haven't noticed that. Should I close this issue(#133)?

alexandremalki, 2017-10-17:
This is done. You can continue on the ticket #131

commit 59b610b6 (2017-10-17): Add new MRJAdapter.jar that fixes #133

I compiled MRJAdapter.jar from github.com/krichter722/mrjadapter with
b1u3-YumaNoguchi fix (#133)
closed 2017-10-17 by alexandremalki