logisim-evolution/logisim-evolution #134: How do I export to VHDL?

smiszym, 2017-10-24:
How can I export the circuit in a form of a generic VHDL - not tied to a particular FPGA board?

So that when I create two gates, XOR and AND, it would just create the following code or similar:

```
entity HALF_ADDER is
  port(A,   B:     in  bit;
       SUM, CARRY: out bit);
end HALF_ADDER;

architecture DATAFLOW of HALF_ADDER is
begin
    SUM   <= A xor B;
    CARRY <= A and B;
end DATAFLOW;
```

maehne, 2018-01-10:
The VHDL code generated by Logisim-evolution is not tied to a particular FPGA board. You can verify this yourself by looking at the generated VHDL files in the Logisim-evolution workspace. However, Logisim-evolution is no synthesis tool. Therefore, it just generates a structural description (a.k.a. netlist) of all circuits in your Logisim file, which instantiate (more or less hard-coded) synthesizable VHDL models of all the used primitive gates and other Logisim components.

Contributions to generate cleaner VHDL or Verilog models are welcome!

I propose that this issue gets closed.

smiszym, 2018-01-10:
Actually, I asked the question, because I didn't know how to export to VHDL at all. I stopped at "Component to FPGA board mapping" - that dialog was unintuitive to me and I gave up.

But now I figured it out. So for anyone interested: how to export to VHDL (quick start):
FPGAMenu > FPGA Commander > (leave options as-is) > Download
and in the dialog, you must select a pin from the list, then select the pin on board, and after all pins you can click "Done".

Btw, why do the files get created in my home directory, under ~/test/main/ ? Can't they be written relative to the circuit's directory, or at least in ~/.logisim or some other hidden directory named after the program?

maehne, 2018-01-10:
It is indeed an annoyance that the FPGA Commander generates its files by default in the home directory. Though, you can easily change that by choosing another directory as the "Workspace" in the FPGA Commander dialog window by clicking on the respective button.

smiszym, 2018-01-10:
This is enough for me. Thanks!

closed 2018-01-10 by smiszym

maehne, 2018-01-10:
I have created an enhancement request (issue #144), which asks to set the default workspace to a more suitable value.

mags0ft, 2024-12-14:
Sorry for opening this closed thread again - it may just be me, but I cannot find the respective option. Maybe I am running an older version?

[image]

maehne, 2024-12-14:
If you select « Synthetisieren », the dialogue will open, which allows to generate HDL code that can be downloaded to an FPGA board.

mags0ft, 2024-12-14:
Ohh, right! Got it now, looks great. Thanks!