logisim-evolution/logisim-evolution #135: ant package fails with JDK version 9.0.1

maehne, 2017-10-28:
Trying to package logisim-evolution using ```ant package``` fails with the new JDK version 9.0.1:
```
$ ant package
Buildfile: ~/Build/logisim-evolution/build.xml

clean:
   [delete] Deleting: ~/Build/logisim-evolution/logisim-evolution.jar
   [delete] Deleting: ~/Build/logisim-evolution/package/Logisim-evolution.jnlp
   [delete] Deleting: ~/Build/logisim-evolution/package/Logisim-evolution_browser.jnlp
   [delete] Deleting: ~/Build/logisim-evolution/package/Logisim-evolution.html

compile:
    [javac] Compiling 1 source file to ~/Build/logisim-evolution/bin
    [javac] warning: [options] bootstrap class path not set in conjunction with -source 1.7
    [javac] 1 warning

jar:
      [jar] Building jar: ~/Build/logisim-evolution/logisim-evolution.jar

package:
  [taskdef] Could not load definitions from resource com/sun/javafx/tools/ant/antlib.xml. It could not be found.

BUILD FAILED
~/Build/logisim-evolution/build.xml:86: Problem: failed to create task or type javafx:com.sun.javafx.tools.ant:deploy
Cause: The name is undefined.
Action: Check the spelling.
Action: Check that any custom tasks/types have been declared.
Action: Check that any <presetdef>/<macrodef> declarations have taken place.
No types or tasks have been defined in this namespace yet

Total time: 3 seconds
```
I am using JDK 9.0.1 on Mac OS 10.13 High Sierra.

maehne, 2017-10-28:
The issue can be fixed by modifying ```build.xml```:
```diff
--- a/build.xml
+++ b/build.xml
@@ -77,7 +77,7 @@
     <target name="package" depends="jar">
         <taskdef resource="com/sun/javafx/tools/ant/antlib.xml"
                  uri="javafx:com.sun.javafx.tools.ant"
-                 classpath=".:${java.home}/../lib/ant-javafx.jar"/>
+                 classpath=".:${java.home}/lib/ant-javafx.jar"/>
         <fx:deploy width="600" height="400"
                    embeddedWidth="100%" embeddedHeight="100%"
                    outdir="${basedir}/package"
```
However, then ```ant``` complains about the unsupported ```basedir``` attribute in the ```fx:platform``` tag:
```
$ ant package
Buildfile: ~/Build/logisim-evolution/build.xml

clean:
   [delete] Deleting: /Users/maehne/Build/logisim-evolution/logisim-evolution.jar

compile:
    [javac] Compiling 1 source file to /Users/maehne/Build/logisim-evolution/bin
    [javac] warning: [options] bootstrap class path not set in conjunction with -source 1.7
    [javac] 1 warning

jar:
      [jar] Building jar: ~/Build/logisim-evolution/logisim-evolution.jar

package:

BUILD FAILED
~/Build/logisim-evolution/build.xml:86: fx:platform doesn't support the "basedir" attribute

Total time: 2 seconds
```
Commenting out the concerned line:
```diff
@@ -91,7 +91,7 @@
                             toolkit="swing"
             />
             <!-- Package only application and dependencies but not the JRE -->
-            <fx:platform basedir=""/>
+            <!-- <fx:platform basedir=""/> -->

             <fx:resources>
                 <fx:fileset dir="." includes="logisim-evolution.jar"/>
```
gets us a bit further. A ```Logisim-evolution.app``` and DMG image is built. However, the App is lacking the Logisim icon and when you start the app bundle, Logisim-evolution is lacking its menu. The pure JAR file does not have that problem: It displays the menu correctly.

maehne, 2017-10-28:
The missing menu bar affects all bundled Java applications on macOS 10.13 High Sierra not using English as system language. E.g., see https://github.com/arduino/Arduino/issues/6548. Until this issue gets fixed by Apple, the workaround is to set the language of Logisim-evolution to English by running the following command in the terminal:
```sh
$ defaults write com.cburch.logisim AppleLanguages '(en)'
```

maehne, 2017-10-28:
That the custom icon isn't added to the Logisim-evolution.app bundle and the ```ant package``` error ```Error: Bundler "DMG Installer" (dmg) failed to produce a bundle.``` while generating the DMG image are confirmed JDK 9 bugs:
https://bugs.openjdk.java.net/browse/JDK-8188763
https://bugs.openjdk.java.net/browse/JDK-8189652

maehne, 2017-10-28:
Just for reference, the current official documentation for using ```ant``` to package Java applications is located at:
https://docs.oracle.com/javase/9/deploy/javafx-ant-tasks.htm#JSDPG177

maehne, 2018-01-10:
The DMG generation failure bugs are consequences of bug [JDK-8190758](https://bugs.java.com/bugdatabase/view_bug.do?bug_id=8190758). The problem is macOS High Sierra specific: The DMG file is created with the wrong file system format (APFS instead of HFS+). The problem has been fixed in JDK 8u162 and the upcoming JDK 10, but not JDK 9. The problem can be worked around by generating only a pkg installer on macOS.

alexandremalki, 2018-02-21:
@maehne  Is the issue solved? if yes please close the issue.

maehne, 2018-02-21:
The issue is not solved. As documented in 52dec05, I currently can only work around the issue by not generating a DMG image, but a PKG installer. We have to wait for the release of JDK 10.

BFH-ktt1, 2018-10-17:
Guys, as JDK 10 is available, can this issue be resolved and closed, or is it still pendent?

maehne, 2018-10-17:
Yes, the bug got fixed in JDK 10 and generating DMG images works now on macOS. However, the situation around `ant package` remains unsatisfying. I did not yet manage to make it work at the same time on JDK 8 and JDK 10 because someone thought it might be a good idea to move around the needed  `ant-javafx.jar` file in the JDK. :-( Now, with JDK 11, they have removed JavaFX from JDK and this probably broke again the packaging process. I will need to find the time for some tests to know more. The results will probably merit a new issue. ;-)

closed 2018-10-17 by maehne