logisim-evolution/logisim-evolution #136: How do i get this to work?

el17js, 2017-10-30:
Its downloaded and so is java but everytime i try to load the command window flashes up for a millisecond and nothing happens

maehne, 2017-10-30:
Your description of your problem is way too vague to give any meaningful response. You need to provide more details:
- What file did you download?
- Which version of Java  did you install? Is it a JRE or JDK? Do other Java programs work in your environment?
- Which platform (OS + version, processor architecture) are you using?
- What file did you try to launch? Give a precise description how to reproduce your problem!

el17js, 2017-10-30:
Sorry my bad.
I'm running windows 10 and I have the latest java. I have deleted and reinstalled at least 5 times today but doesn't work. It's a new laptop so I don't have any other java apps on it. My mates laptop has it running fine but for some reason every time I double click logisim evolution it just briefly flashes up command window and disappears again.

maehne, 2017-10-30:
Your new message contains few extra information. We only know that you experience the problem on Windows 10. You did not answer precisely enough the other questions. What do you mean with “the latest Java version”? Java 9 JRE version 9.0.1? In that case it is very likely that you are affected by issue #131, which has beeen fixed in the source code. However, there is no new binary release yet. Either stick to Java 8 or build logisim from source (requires JDK and Apache Ant to be installed).

maehne, 2018-01-10:
I suggest that this issue is closed due to lack of feedback from the reporter of the issue.

closed 2018-02-13 by BFH-ktt1